CloverHackyColor/CloverBootloader #368: Catalina cannot be used on Clover 5130.

kwmlman, 2021-02-24:
[Image-01]
[image]

[Image-02]
[image]

Clover 5123-5128 worked well for both Catalina and Big Sur,
In Clover 5129-5130, Big Sur works fine, but Catalina does not work because of [in memory panic stackshot succeeded].

**Is there any way this can be solved?**

kwmlman, 2021-02-24:
(( My System))

MSI B360M Mortar / i5-9400F / Sapphire rx570 4GB ( 1660 Ti 6GB )

kwmlman, 2021-02-24:
[Image-03]
[image]

Just removing the SSDT-PLUG-ME.aml in the patched folder solved this problem. 
This solution worked on clover 5129+5130+5131.

Clover Team's hopes will be accomplished wonderfully... (^-^)

SergeySlice, 2021-02-25:
Well but the SSDT-PLUG-ME has no relation to Clover team.

closed 2021-02-25 by SergeySlice