CloverHackyColor/CloverBootloader #369: r5130 stopped booting on esxi

kcj83, 2021-02-27:
5129 working perfectly, no issue at all on esxi-osx-vm. as soon as i upgrade to 5130 or newer, the loader crashes/shuts down the vm right after boot selection menu. could be due to the datahubdxe embedded driver causing issues with the vm/esxi hypervisor. please fix/revert this if possible?! let me know if i can provide anything to help. @SergeySlice

SergeySlice, 2021-02-27:
It is not a news that Clover N+1 is not working while Clover N works fine.
May be you have to wait a version 5132 which will not work while 5131 will be perfect?
Anyway can you provide any proof of your trouble?

kcj83, 2021-02-27:
both 5130 and 5131 are NOT working, while 5129 and below are working just fine.  What changes have been made that would negatively impact booting osx vm from esxi hypervisor in this way? the only thing ive seen thus far is the datahubdxe embedded driver changes... and there is no way to disable this driver since it is now embedded directly, correct?

SergeySlice, 2021-03-07:
Some logs will be more informative.

kcj83, 2021-03-08:
Of course Sergey, which logs would you like? Or can I just attach the entire EFI Clover directory?

papasax, 2021-03-09:
I don't know if this is related, but since r5130, my MacOS entry just disappeared from the clover boot menu. I can still boot linux however. I had to revert to r5128 to get my hackintosh running again (did not try r5129).

SergeySlice, 2021-03-16:
Both Clover 5130 and 5131 works showing macOS entry. 
Provide, please logs, screenshot or other proof of your observations.

closed 2021-03-16 by SergeySlice

kcj83, 2021-03-16:
please dont close this, i will provide logs ASAP.