CloverHackyColor/CloverBootloader #370: Cannot upgrade from 5123.1 to 5130/5131 due to kernel panic

jedis00, 2021-02-28:
Hello,

I am currently running Sierra and was using Clover 4586. Ran into issues upgrading to the latest, so slowly and meticulously upgraded as far as I can go, solving problems with drivers along the way. Been at this all day and appreciate any help.

Upgraded from every release in succession:
4586 -> 4700
4700 -> 4813
4813 -> 4895
4895 -> 4920
4920 -> 4945
4945 -> 4988
4988 -> 5018
5018 -> 5045
5045 -> 5058
5058 -> 5070
5070 -> 5089
5089 -> 5094
5094 -> 5099
5099 -> 5101
5101 -> 5115
5115 -> 5119
5119 -> 5120
5120 -> 5123.1 **Successful**
5123.1 -> 5130 **Failed**
5123.1 -> 5131 **Failed**

There is not any pkg release installers after 5123.1 to 5130. I tried downloading 5127 and extracting CLOVERX64.efi.zip over top of EFI/CLOVER/CLOVERX64.efi, however, it freezes at the Apple logo with 5127. With no errors shown.

With 5130 and 5131, it kernel panics with the following: https://i.imgur.com/Dci6GBZ.jpg

Any ideas?

stevezhengshiqi, 2021-02-28:
Hi, please upload your CLOVER folder for better diagnose. A debug log is also required.

When updating from r5123, you also need to update your config.plist by following https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist. The `Quriks` section needs to be properly configured.

jedis00, 2021-03-01:
@stevezhengshiqi Thank you. Here is the logs for trying to go from 5123.1 to 5130 (the first release that still has a pkg installer).

Here's my [debug.log](https://www.dropbox.com/s/skngc9km3dib98y/2021-3-1_0-10-53_CLOVERX64.efi.log?dl=0).

Here's my [CLOVER ](https://www.dropbox.com/s/s3tyojgq1vb80ac/EFI-clover_r5130.zip?dl=1)directory.

kushwavez, 2021-03-01:
In the kernel panic it is clearly showing the problem. BrcmPatchRAM2. 
Update all of your kexts to the latest. 

Also, you have FakePCIID_Broadcom_WiFi.kext, and FakePCIID, that is deprecated. Use AirportBrcmFixup.kext instead. 
BrcmFirmwareRepo is bad, use BrcmFirmwareData. 
Shiki, NvidiaGraphicsFixup not needed, use WhateverGreen.kext instead
Also try to use VirtualSMC instead of FakeSMC (I suppose FakeSMC is also really outdated in your case), delete FakeSMC and SMCHelper.efi then use VirtualSMC.kext

jedis00, 2021-03-02:
@kushwavez Is that how to interpret the screenshot?

Why does the latest Clover updates start causing this to happen? How do you know which kexts replace others? I’m about to upgrade my Hackintosh from Sierra to High Sierra and beyond, so any info you can provide would be greatly appreciated. Thank you.

kushwavez, 2021-03-02:
> Is that how to interpret the screenshot?

Yes. 
It says Kernel Extensions in backtrace... 
After that it says BrcmPatchRAM2.kext. Meaning that is the cause. 

> Why does the latest Clover updates start causing this to happen?

I don’t know about that. 

> How do you know which kexts replace others?

FakePCIID_Broadcom_WiFi is an old kext, AirportBrcmFixup is basically an updated replace for that. 
Same for NvidiaGraphicsFixup and Shiki. they’re merged into WhateverGreen.kext a year ago(?)

papasax, 2021-03-09:
I don't know if this is related, but since r5130, my MacOS entry just disappeared from the clover boot menu. I can still boot linux however. I had to revert to r5128 to get my hackintosh running again (did not try r5129).

SergeySlice, 2021-03-21:
There is no Clover issue. There is user's mistake.

closed 2021-03-21 by SergeySlice

papasax, 2021-03-21:
I use the newest drivers. I have the latests ApfsDriverLoader.efi and VboxHfs.efi loaded. My drivers are not in drivers/off/UEFI but in drivers/UEFI (I don't have a directory named drivers/off, is it mandatory?). It still does not recognize my mac entry. r5128 does.