CloverHackyColor/CloverBootloader #376: Clover disregards GUI Custom Tool Argument

LeeBinder, 2021-03-08:
Issue occurs with latest Clover 5131 - previous versions not tested.

```
<dict>
	<key>Volume</key>
	<string>\</string>
	<key>Path</key>
	<string>\EFI\Clover\Tools\ControlMsrE2.efi</string>
	<key>Arguments</key>
	<string>unlock</string>
	<key>FullTitle</key>
	<string>1-ControlMsrE2</string>
	<key>Hotkey</key>
	<string></string>
	<key>Hidden</key>
	<false/>
	<key>Disable</key>
	<false/>
	<key>Ignore</key>
	<false/>
</dict>
```
`ControlMsrE2.efi` runs but complains that an argument is required but missing. In OC the same thing works. The parameter must follow `ControlMsrE2.efi` without a preceding `-` only divided by a space, like this:

`ControlMsrE2.efi unlock`

I also tried `<string>"unlock"</string>` and `<string>'unlock'</string>`
and also
`<string> unlock</string>` , `<string> "unlock"</string>` and `<string> 'unlock'</string>`

Here's the .zip with everything you need so you can replicate the bug in your setup: [GUI Custom Tool Arguments disregarded.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6104765/GUI.Custom.Tool.Arguments.disregarded.zip)

LeeBinder, 2021-03-10:
tested all the way down to r5108 - same thing. Must be an old bug.

LeeBinder, 2021-03-20:
source code (thanks to cecekpawon for the pointer): https://github.com/zhen-zen/OpenCorePkg/tree/ControlMsrE2/Application/ControlMsrE2

LeeBinder, 2021-03-21:
on Mar 21, 2021, 7:21 AM GMT+1 @SergeySlice [wrote](https://github.com/CloverHackyColor/CloverBootloader/issues/377#issuecomment-803520230):
> ControlMsrE2.efi works as well in Clover 5131 with Shell provided with the Clover.
> [image]

yes, it works fine if run via shell. What happens when you run it from `Tools` on your computer - does it work from there for you, either with any of the 3 arguments
```
<dict>
	<key>Volume</key>
	<string>\</string>
	<key>Path</key>
	<string>\EFI\Clover\Tools\ControlMsrE2.efi</string>
	<key>Arguments</key>
	<string>unlock</string>
	<key>FullTitle</key>
	<string>1-ControlMsrE2</string>
	<key>Hotkey</key>
	<string></string>
	<key>Hidden</key>
	<false/>
	<key>Disable</key>
	<false/>
	<key>Ignore</key>
	<false/>
</dict>
```
or even with empty `<key>Arguments</key>` `<string></string>`?

jief666, 2021-03-22:
Should work now with the latest commit, or this efi : [CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6184373/CLOVERX64.efi.zip)

Please test. If works, close this issue please.

LeeBinder, 2021-03-22:
@jief666 sorry but no, error persists here. Additionally I even just tried the [latest ControlMsrE2.efi compile](https://github.com/acidanthera/OpenCorePkg/suites/2312082823/artifacts/48544825), both release and debug version.

I verified Clover really did boot from the `CLOVERX64.efi` from your link, renamed and copied to `EFI/BOOT/BOOTX64.efi`:

[image]

I even tried the trick which works with Brummbär's original CFGlock.efi - as an intermediate step first open my Laptop's UEFI BIOS boot menu, but still no-go.

Does `ControlMsrE2.efi` work for you via tools? If so, maybe you could zip and attach your EFI folder - (minus) themes and SMBIOS data in config.plist.

jief666, 2021-03-22:
Send your debug.log...

LeeBinder, 2021-03-22:
sure - can do tomorrow - have to leave right this second

LeeBinder, 2021-03-23:
> Send your debug.log...

here you go -> [2021-3-23_20-27_BOOTX64.EFI.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6192259/2021-3-23_20-27_BOOTX64.EFI.log.zip)

also preboot.log, for what it might be worth -> [preboot.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6192274/preboot.log.zip)

jief666, 2021-03-23:
That is the log for when you boot macOS. I'd like the log when you tried to boot ControlMsrE2.efi.

jief666, 2021-03-23:
Also, in that log, I don't see any custom tool "ControlMsrE2.efi".

LeeBinder, 2021-03-23:
> Also, in that log, I don't see any custom tool "ControlMsrE2.efi".

[image]

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6192550/config.plist.zip)

> That is the log for when you boot macOS. I'd like the log when you tried to boot ControlMsrE2.efi.

this time I invoked both via Clover > tools, `ControlMsrE2.efi` and `ControlMsrE2_debug.efi` while Clover > Boot > Debug is enabled - hopefully efi tools log into that debug log, too.

[2021-3-23_21-18_BOOTX64.EFI.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6192565/2021-3-23_21-18_BOOTX64.EFI.log.zip)

jief666, 2021-03-23:
Thanks, your config.plist and log were helpful.
I didn't realize it was a custom tool and not a custom entry.
Should work now. Compile last commit or : [CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6192817/CLOVERX64.efi.zip)

LeeBinder, 2021-03-23:
> I didn't realize it was a custom tool and not a custom entry.

my bad - should've set the issue title to double-size and red .. ;)

> Should work now. Compile last commit or : CLOVERX64.efi.zip

YES, you ROCK as usual - finally the arg is being passed over - great job!

Clover still needs the UEFI BIOS boot menu to be invoked first to initiate the MSR E2 register so that both, `ControlMsrE2.efi` and `CFGLock.efi` find it if called via Clover, but oh well, that's a minor oddity in contrast to OC. Fair enough/ fine for now.

Thanks again @jief666 for solving this Bug!

closed 2021-03-23 by LeeBinder

jief666, 2021-03-24:
> my bad - should've set the issue title to double-size and red .. ;)

Of course NOT. I never used Custom tool and I misinterpreted it as a tool configured as a custom entry. My bad.

> Clover still needs the UEFI BIOS boot menu to be invoked first

Not sure to understand that. If you'd like an improvement, it maybe possible...

LeeBinder, 2021-03-24:
> Not sure to understand that. If you'd like an improvement, it maybe possible...

I explained it [here](https://github.com/LeeBinder/Asus-Vivobook-S510UA-Hackintosh/blob/main/README.md/#user-content-msr-unlock-clover)

jief666, 2021-03-25:
> I explained it [here](https://github.com/LeeBinder/Asus-Vivobook-S510UA-Hackintosh/blob/main/README.md/#user-content-msr-unlock-clover).
What you explained here is now possible again, right ?

> Clover still needs the UEFI BIOS boot menu to be invoked first
I don't know what means : Clover finds UEFI Bios boot menu.

> w/o detour
No idea what is that "detour".

> What about on our hack if you run `ControlMsrE2.efi` from `Tools` in Clover?
Something like : "register not found". I'm not even sure what is MsrE2, but I don'thave that, I think.

LeeBinder, 2021-03-25:
jief, thanks for being willing to dig even deeper into this, but if all this does not tell you anything, it's OK to just forget about it - it's fine enough now the way it is after your corrections. The extra step with having to enter the UEFI boot menu before running Clover for MSR E2 register detection and flipping the bit might just be an Asus Vivobook S15 oddity, so we can simply let it slide for now. I'd say it might help more Clover users to rather focus on the other issues here :)