CloverHackyColor/CloverBootloader #377: [REGRESSION] r5131 breaks CFGLock.efi by Brumbaer - fine with r5130

LeeBinder, 2021-03-11:
[image]

after replacing 5131 `EFI/BOOT/BOOTX64.efi` with the previous 5130 `BOOTX64.efi`, functionality is back to what it's supposed to be:
[image]

[my sample Clover EFI with CFGLock.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6122617/my.sample.Clover.EFI.with.CFGLock.efi.zip)

Let me know if you need any logs

Link to topic @ insanelymac: [CFGLock - unlock (MSR 0xE2)](https://www.insanelymac.com/forum/topic/344035-cfglock-unlock-msr-0xe2/) by Brummbär

mackonsti, 2021-03-14:
Hi @LeeBinder you are launching this via an interactive UEFI Shell correct?

I have downloaded the latest CFG Unlock tool from https://github.com/utopia-team/opencore-debug/releases and it works fine with OpenCore 0.6.6 just tested, it as I just updated my NUC10 BIOS and MSR was re-locked. Perhaps worth trying it?

Brumbaer's (excellent) work was taken over by this team, it seems.

LeeBinder, 2021-03-14:
@mackonsti 

> Hi @LeeBinder you are launching this via an interactive UEFI Shell correct?

No - it's not supposed to be necessary to do so, esp. with Brummbär's original CFGlock.efi because it does not even use (let alone require) any params to be added interactively. It's like: the bug is that the car does not drive forward anymore, so let's drive it backwards via reverse gear, so at least it runs .. that's not how to solve an issue ;)

> I have downloaded the latest CFG Unlock tool from https://github.com/utopia-team/opencore-debug/releases
> [..]
> Brumbaer's (excellent) work was taken over by this team, it seems

yes, OC debugs's "utopia team" took over Brummbär's `CFGlock.efi` code, re-labeled it `ControlMsrE2.efi` and made it interactive, requiring one of three arguments to be passed over. Unfortunately not even that is working in Clover, see Issue #376 :(

I'm confident team Clover will manage to at least get one of the MSR E2 unlock tools working (again) 🥇

LeeBinder, 2021-03-14:
What's needed is some other users who test this on their hack and report back if they can get CFGlock.efi to work via Tools in Clover 5131, and on what hardware + BIOS.

SergeySlice, 2021-03-18:
There is no CFGLock sources so I can't check what is happen.
Is there any problem with an open-source project?

LeeBinder, 2021-03-18:
> There is no CFGLock sources so I can't check what is happen.

unfortunately you're right

> Is there any problem with an open-source project?

sorry, I don't understand your question. Can you please re-phrase what you mean?`ControlMsrE2.efi` does not appear to be open-source either (Issue #376)

SergeySlice, 2021-03-19:
Some similar observation with other efi application? Which can be controlled.

LeeBinder, 2021-03-19:
None I'm aware of.

BTW, I got it working with the latest Clover with a trick:

- right after turning on or rebooting my VivoBook, press the ESC key to intercept booting and to enter the built-in Boot Menu
- chose the partition with Clover, in my case the ESP
- run CFGLock.efi via Tools

BINGO, working!

Somehow this initiation via the boot menu is necessary.

cecekpawon, 2021-03-20:
@SergeySlice I hope this point to the right sources, please check: Active [pull-request](https://github.com/acidanthera/OpenCorePkg/pull/167) | [Original source](https://github.com/zhen-zen/OpenCorePkg/tree/ControlMsrE2/Application/ControlMsrE2) under review.

LeeBinder, 2021-03-20:
thank you @cecekpawon for your link to the source - I'll also add to #376

SergeySlice, 2021-03-21:
Anyway I didn't confirm the title problem
[image]

SergeySlice, 2021-03-21:
ControlMsrE2.efi works as well in Clover 5131 with Shell provided with the Clover.
[image]

closed 2021-03-21 by SergeySlice

LeeBinder, 2021-03-21:
> Anyway I didn't confirm the title problem

yes, `CFGLock.efi` works fine if run via shell even with r5131.

Even though you closed this issue, I re-labeled the title more appropriately to "_[REGRESSION] r5131 breaks CFGLock.efi (by Brumbaer) if run via 'Tools' entry - fine with r5130_".

@SergeySlice what happens when you run `CFGLock.efi` from `Tools` on your computer? Does it work from there for you yes/no?

`ControlMsrE2.efi`: I'll shift the conversation about it over to #376

jief666, 2021-03-27:
Hi, is this really solved ? Or just not needed ?
Because the problem depicted in #376 wasn't new. So if it was working on 5130, there must be something else...

LeeBinder, 2021-03-27:
CFGLock is outdated, and as per quote slice from here re. ControlMsrE2:

> The application is now compiled with Clover and will be included into next Clover release. 

I'd say let's wait and see how that turns out to be. From my side we can forget about CFGLock and anything related to it.

jief666, 2021-03-28:
I still like to find regressions (and not only because I'm the one to cause them:-) ! Another tool maybe broken by that.
I'll integrate your "config-custom-tools_sample.plist" and compare. I'll post the result here.
Thanks for your "my.sample.Clover.EFI.with.CFGLock.efi.zip".

jief666, 2021-03-28:
I tried with 5130 and it said "Can't find any Variable with cfg in name.
As you said, let's forget about that. If an other tool doesn't work, let's open a new issue.

LeeBinder, 2021-03-28:
Okee-doke - seems like the most economic decision for now.

jief666, 2021-03-28:
Economic !!! Yes, exactly right. And I need it :-)
Thanks for your understanding.