CloverHackyColor/CloverBootloader #378: [Enhancement] Use RenameDevices for Methods too

kushwavez, 2021-03-11:
So what I tried to do is rename `_GPE.NTFY -> XTFY`, and `_SB.PCI0.RP09._INI -> XINI` in DSDT
`_GPE.NTFY`:
- _GPE (Scope)
- NTFY (Method)

[image]

Looked at he log, I see it is not renamed. 
```
[...]
0:712  0:000  NTFY:_GPE:->will be renamed to XTFY
[...]
4:851  0:003  Name: NTFY, Bridge: _GPE, Replace: XTFY
4:852  0:001    0 replacements
[...]
```
So according to @SergeySlice it's normal because ONLY Devices can be renamed, Methods cannot.

But maybe because of a bug, if there is a **Device** after the **Scope**, the **Method** can be renamed too.
Example what I mentioned: `_SB.PCI0.RP09._INI -> XINI`
- `_SB.PCI0` (Scope)
- `RP09` (Device)
- `_INI` (Method)

**Result:** 
```
[...]
4:250  0:000  Name: _INI, Bridge: RP09, Replace: XINI
4:253  0:003    1 replacements
[...]
```
[image]
As you can see, it is renamed.

**Another example:** `SB.PCI0.RP09._PS3 -> XPS3` in **SSDT-4**
- SB.PCI0.RP09 (Scope)
- _PS3 (Method)

(so there is no device after Scope, so it'll fail)
**Result:** 
```
0:742  0:000  _PS3:RP09:PCI0:_SB_:->will be renamed to XPS3
[...]
7:700  0:001  Name: _PS3, Bridge: RP09, Replace: XPS3
7:705  0:003    0 replacements
```
Now I use Bin rename patches to do this, but for the ONLY and EXACT match I need to find that one Method with HexFiend in DSDT.aml, and find a pattern that matches ONLY that exact one Method that needs to be patched.

For example if I want to patch `_SB.PCI0.RP09._INI -> XINI` without using RenameDevices I have to find the pattern in HexFiend, that matches only `_INI` that is inside `_SB.PCI0` Scope, `RP09` Device. But the problem is that for `_INI` there are `44` matches in DSDT.aml, so it's really hard to find that only match. As for now,  I'm lucky because the `_SB.PCI0.RP09._INI -> XINI` RenameDevice worked (because of a bug), so I could find it with HexFiend, and use rename patch : 

**Find:** `5F414452 00A00E92 93525041 390A00A4 52504139 A107A40C 00001D00 084C5452 5A0A0008 4F42465A 0A00084C 4D534C0A 00084C4E 534C0A00 142A5F49 4E49`
**Replace:** `5F414452 00A00E92 93525041 390A00A4 52504139 A107A40C 00001D00 084C5452 5A0A0008 4F42465A 0A00084C 4D534C0A 00084C4E 534C0A00 142A5849 4E49`
As you can see it is really long, because it is the only pattern that matches with that one `_INI` Method

It would be convenient to use RenameDevices to Rename Methods too under Scopes.

SergeySlice, 2021-04-15:
It works
~~~~
            <key>RenameDevices</key>
            <dict>
                <key>_SB.GINF</key>
                <string>XINF</string>
            </dict>

~~~~

And the result
~~~~
    Scope (_SB)
    {
        Method (XINF, 2, NotSerialized)
        {
            If ((PCHS == SPTL))
            {
                Return (DerefOf (DerefOf (GPCL [Arg0]) [Arg1]))
            }
            Else
            {
                Return (DerefOf (DerefOf (GPCH [Arg0]) [Arg1]))
            }
        }

~~~~

kushwavez, 2021-04-16:
hmm unfortunately in my case it doesn't work.

**config.plist:**
```
<key>RenameDevices</key>
		<dict>
			<key>_SB.PCI0.RP09._INI</key>
			<string>XINI</string>
			<key>_GPE.NTFY</key>
			<string>XTFY</string>
			<key>_SB.PCI0.RP09._PS3</key>
			<string>XPS3</string>
			<key>_SB.PCI0.RP09._PS0</key>
			<string>XPS0</string>
			<key>_SB.PCI0.XHC.RHUB._PS0</key>
			<string>XPS0</string>
			<key>_SB.PCI0.XHC.RHUB._PS3</key>
			<string>XPS3</string>
		</dict>
```

**bootlog.log:**
```
1:210  0:002  '_INI:RP09:PCI0:_SB_' -> will be renamed to 'XINI'
1:212  0:002  'NTFY:_GPE' -> will be renamed to 'XTFY'
1:215  0:002  '_PS3:RP09:PCI0:_SB_' -> will be renamed to 'XPS3'
1:218  0:002  '_PS0:RP09:PCI0:_SB_' -> will be renamed to 'XPS0'
1:221  0:002  '_PS0:RHUB:XHC_:PCI0:_SB_' -> will be renamed to 'XPS0'
1:223  0:002  '_PS3:RHUB:XHC_:PCI0:_SB_' -> will be renamed to 'XPS3'
```
```
6:124  0:006  Name: _INI, Bridge: RP09, Replace: XINI
6:130  0:005  Name: NTFY, Bridge: _GPE, Replace: XTFY
6:134  0:003  Name: _PS3, Bridge: RP09, Replace: XPS3
6:138  0:004  Name: _PS0, Bridge: RP09, Replace: XPS0
6:142  0:004  Name: _PS0, Bridge: RHUB, Replace: XPS0
6:147  0:004  Name: _PS3, Bridge: RHUB, Replace: XPS3
6:151  0:004    3 replacements
```

**What are renamed:**
DSDT: `_SB.PCI0.RP09._INI` to `XINI`
```
Scope (\_SB.PCI0)
    {
 Device (RP09)
        {
            Method (XINI, 0, NotSerialized)
            {
                LTRZ = LTR9 /* \LTR9 */
                LMSL = PML9 /* \PML9 */
                LNSL = PNL9 /* \PNL9 */
                OBFZ = OBF9 /* \OBF9 */
            }
     }
```

DSDT: `_SB.PCI0.XHC.RHUB._PS0` to `XPS0`
```
 Scope (\_SB.PCI0)
    { 
Device (RHUB)
            {
                Method (XPS0, 0, Serialized)
                {
                    If ((\_SB.PCI0.XHC.DVID == 0xFFFF))
                    {
                        Return (Zero)
                    }

[... 188 lines cut ...]
SSDT-3-RVP7Rtd3: `_SB.PCI0.RP09._PS3` to `XPS3`
```
Scope (\_SB.PCI0.RP09)
        {
          Method (_PS3, 0, Serialized)  // _PS3: Power State 3
            {
                OperationRegion (PLTR, SystemMemory, PWRM, 0x0800)
                Field (PLTR, AnyAcc, NoLock, Preserve)
                {
                    Offset (0x3EC), 
                    Offset (0x3EE), 
                    BI16,   1, 
                    Offset (0x3EF), 
                    BI24,   1
                }

                BI16 = 0x00
                BI24 = 0x00
            }
        }
```

Attaching them:

[SSDT-DSDT.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6323842/SSDT-DSDT.zip)