CloverHackyColor/CloverBootloader #379: Binpatch: "bin not found / already patched" if the pattern is in SSDT, but it's being patched even so

kushwavez, 2021-03-11:
If I rename something that is not in DSDT, but in SSDT, log says "bin not found / already patched" even if it's being patched.

For example` _UPC to XUPC` (_UPC is ONLY in SSDT's, not in DSDT):
 I looked up my SSDT-4.aml with HexFiend to find the pattern, found:

- _UPC? (5F555043 08A4)
- 6_UPC (365F5550 43)
- %_UPC (255F5550 43)

If I look at the log, it says:
```
10:786  0:000   - [_UPC? to XUPC]: pattern 5F555043, bin not found / already patched!
10:786  0:000   - [6_UPC to XUPC]: pattern 365F5550, bin not found / already patched!
10:786  0:000   - [%_UPC to XUPC]: pattern 255F5550, bin not found / already patched!
```
In this case this msg is at the first attempt for patch, so it isn't "already patched"
 
But they're patched, looking at my SSDT-4, so that's why I say it's cosmetic issue:
[image]

If it's in my DSDT, for example _GPE.NTFY (CGNTFY: 43474E54 4659) to XTFY it says:
`10:787  0:000   - [GPE_NTFY to XTFY]: pattern 43474E54, patched at: [ (16B81) ]`
So, if it's in DSDT it's reporting correctly.

Attaching log:
[2021-3-11_11-23-21_CLOVERX64.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/6122951/2021-3-11_11-23-21_CLOVERX64.efi.log)

SergeySlice, 2021-03-28:
The message is always right. There must be other reason why the SSDT was already patched.

kushwavez, 2021-03-28:
But if I disable the rename patch and then grab the running SSDT I can see it is not renamed.

SergeySlice, 2021-03-29:
I have other experience
~~~~
118:253  0:133  Patching SSDTs: 9 patches each
118:345  0:091   pattern 5F44534D, patched at: [ (40D) ]
118:845  0:499   pattern 5844534D, bin not found / already patched!
119:028  0:183   pattern 48454349, bin not found / already patched!
119:253  0:225   pattern 47465830, bin not found / already patched!
119:495  0:241   pattern 50454750, patched at: [ (233) ]
119:911  0:416   pattern 414C5344, bin not found / already patched!
120:136  0:225   pattern 53415430, bin not found / already patched!
~~~~

kushwavez, 2021-03-29:
oh my god I'm such an idiot. 

You're right. I just don't know how I missed the line "Patching SSDTs" . That is the reason! Obviously it is saying that because it is in "Patching DSDT".

My mistake. Sorry for wasting your time.

closed 2021-03-29 by kushwavez