CloverHackyColor/CloverBootloader #380: trouble updating : macOs Installer entry isn't diplayed

xma, 2021-03-14:
While trying to upgrade from 11.2 to either 11.2.1 , 11.2.2 or 11.2.3
* From macOS "Software update" : everything is written and computer reboot as expected
* On Clover, there is no entry related to "macOS Installer" -> Selecting regular hard drive ends in previous operating system i.e: not updated

Workaround was to use OpenCore : once update is applied, an entry "macOS Installer" is seen at boot, update complete w/o trouble.

SergeySlice, 2021-03-16:
Not true. Clover 5131 successfully updates Big Sur and the Installer entry shown at first reboot.

xma, 2021-03-16:
that wasn't the case, 

I was using clover 5131 at the time (commited on my repo 23 feb https://github.com/xma/T480-Clover/commit/1b447be6b88e16d6186825a3fd0cc7646bdb0822)

10 & 11 march : I've updated to 11.2.2 from 11.2.1   and then to 11.2.3

Next time there's a minor version (11.3?) I'll post screenshof of Clover _and_ OC : you'll see the difference

kushwavez, 2021-03-25:
@SergeySlice He is right, I'm facing the same issue, trying to install a Catalina 10.15.7 supplemental patch, but the Installer entry isn't there so I can't boot off from it. 
Clover v5130: same
Clover v5129: can't test, I can't find the .efi
Clover v5128: OK, the entry is there and I can boot from it

[2021-3-25_21-47_CloverX64-RELEASE_XCODE8-20210322154001-829fd04.efi.log](https://github.com/CloverHackyColor/CloverBootloader/files/6208090/2021-3-25_21-47_CloverX64-RELEASE_XCODE8-20210322154001-829fd04.efi.log)
[preboot_v5128.log](https://github.com/CloverHackyColor/CloverBootloader/files/6208091/preboot_v5128.log)

kushwavez, 2021-03-28:
@xma fixed in v5132

closed 2021-03-29 by SergeySlice