CloverHackyColor/CloverBootloader #382: The Model Identifier is not correct in Clover 5132

rex-lapis, 2021-03-30:
Hi Clover team,
First at all, thank you for your hard work for Clover 5132 version.
I found a small issue with model identifier after upgrading to Clover 5132. The model identifier is not correct. My SMBIOS is MacBookPro15,1 but it shows MacBook only in System Infomation, also in the Hackintool.
Hope you will fix it soon.
[image]
[image]

kushwavez, 2021-03-31:
hmm mine is okay, but I set the SMBIOS using Clover Configurator. example:
[image]
[image]

Did you set SMBIOS info manually or automatically by Clover?

kushwavez, 2021-03-31:
@rex-lapis If the fields are set in Clover Configurator -> SMBIOS by the two arrows and select (or clicking generate), then it's not automatic, but manual. It's just Clover Configurator generating a new SMBIOS.

rex-lapis, 2021-03-31:
> @rex-lapis If the fields are set in Clover Configurator -> SMBIOS by the two arrows and select (or clicking generate), then it's not automatic, but manual. It's just Clover Configurator generating a new SMBIOS.

hmm... i will try. Thank you

rex-lapis, 2021-03-31:
@kushwavez seems not work for me... when Clover 5131 everything is fine. Also in my Clover Configurator, I cant click to arrow button to select SMBIOS, seems like it's broken or freeze, I dont know @.@

closed 2021-04-06 by rex-lapis