CloverHackyColor/CloverBootloader #383: Clover unable to find the Haiku install

extrowerk, 2021-04-06:
I just extracted the latest Clover release (5312) to my EFI partition, without any change in the config file.
Clover starts successfully on my T440s (UEFI, without CSM), but it doesn't lists my Haiku install.
AFAIK Clover should list automatically any available bootable disk. Also AFAIK Clover supposed to support Haiku.

SergeySlice, 2021-04-11:
I need more information what to boot in the case of Haiku.

extrowerk, 2021-04-11:
I can provide more info if you specify what you need.

extrowerk, 2021-04-11:
Haiku can be booted in legacy and in UEFI mode. My install can be booted woth both way, but Clover doesnt list any with the default config.
Does Clover supposed to discover bootable drives automatically?

jief666, 2021-04-20:
Yes. Could you send a zip of a clean EFI haiku folder. We need to know where and what name is the boot loader.

extrowerk, 2021-04-20:
I can do that! Should i upload it to Github in the comment?

jief666 írta 2021. ápr.. 20, K-n 14:05 órakor:

extrowerk, 2021-04-20:
But do i understand correctly you are planing to support only EFI based Haiku installs (so no MBR)?

Mizsei Zoltán írta 2021. ápr.. 20, K-n 14:18 órakor:

jief666, 2021-04-21:
I don't know why legacy doesn't work. I never looked in the legacy code in Clover. I will.
But for now, let's do EFI first.
You can use a transfer service like wetransfer and post the link. At your convenience.

extrowerk, 2021-04-21:
[HAIKU_R1_beta2_EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6350300/HAIKU_R1_beta2_EFI.zip)

Here is the ESP partition content from Haiku R1b2. Hope it will help. If you need a dd dump or something let me know.

extrowerk, 2021-04-21:
And here is the first 512 bytes dd dump of the Haiku partition.

[haiku_partition_512_bytes_dd_dump.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6350350/haiku_partition_512_bytes_dd_dump.zip)

jief666, 2021-04-22:
1st : for EFI boot.
You'll to create a custom entry in Clover GUI section :
```
	<key>GUI</key>
	<dict>
		<key>Custom</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Title</key>
					<string>haiku</string>
					<key>Path</key>
					<string>\EFI\Haiku\BootX64.efi</string>
					<key>Type</key>
					<string>other</string>
					<key>TextMode</key>
					<true/>
				</dict>
			</array>
		</dict>
	</dict>

```
You can choose the path of the efi file and even rename the efi file, of course.
NOTE : I had to create the option "<key>TextMode</key>", so that will work only with latest Clover.
Here is a version : 
[CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6357966/CLOVERX64.efi.zip)

Let me know if that works.

For legacy boot : do you really need that in 2021 :-) ?

extrowerk, 2021-04-22:
Thanks you! Seems to be working fine.

[config.plist](http://0x0.st/-mH2.plist)
[preboot.log](http://0x0.st/-mH_.log)

extrowerk, 2021-04-22:
[image]

extrowerk, 2021-04-22:
AFAIR legacy boot were working before the PackageManagement introduction in Haiku and the infrastructure for Clover to recognize the Haiku legacy disks are there just some adjustments would be needed (this is an assumption). It would be nice if it could be fixed.

extrowerk, 2021-04-22:
Your patched efi file booted Haiku successfully in EFI mode.

jief666, 2021-04-22:
And for Legacy, try this : 
[CloverX64-2021-04-22-21-56-21-a1a27d2-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6360639/CloverX64-2021-04-22-21-56-21-a1a27d2-dirty-jief.zip)

extrowerk, 2021-04-22:
> And for Legacy, try this :
> [CloverX64-2021-04-22-21-56-21-a1a27d2-dirty-jief.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6360639/CloverX64-2021-04-22-21-56-21-a1a27d2-dirty-jief.zip)

I tried to boot this in UEFI mode on, CSM disabled: got black screen

jief666, 2021-04-22:
You mean you didn't get the Clover gui ?

jief666, 2021-04-22:
What about this : [CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6361409/CLOVERX64.efi.zip) ?

extrowerk, 2021-04-23:
> You mean you didn't get the Clover gui ?

Correct.

jief666, 2021-04-23:
No GUI even with the last CloverX64.efi I've sent ?
Could activate the debug log in the boot section of config.plist ?
```
        <key>Boot</key>
        <dict>
            <key>Debug</key>
            <true/>
        </dict>

```

extrowerk, 2021-04-23:
> What about this : [CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6361409/CLOVERX64.efi.zip) ?

Not tested yet. I am working right now.

jief666, 2021-04-23:
Sure, no problem. Whenever.

extrowerk, 2021-04-23:
> > You mean you didn't get the Clover gui ?
> 
> Correct.

Sorry, my mistake, gave a wrong name for the efi file, scratch everything. So your first legacy enabled clover build shows up, Legacy partition detected successfully, but starting it results a black screen and thats all.

Second legacy-enabled one not tested yet.

extrowerk, 2021-04-23:
Second one finds Haiku-legacy too, but attempting to booting it results only in a black screen with a blinking prompt.
Maybe my strange (and unsupported) hybrid install is the culprit. I don't know. Please forget about the legacy boot now, but please do commit your efi boot changes in master if possible. Thanks for your hard work!

SergeySlice, 2021-05-15:
The issue resolved. If other problem exists then open new issue.

closed 2021-05-15 by SergeySlice