CloverHackyColor/CloverBootloader #387: Freezing

Savio-Brazil, 2021-04-20:
The latest release of Clover (r5133) is freezing right on startup on my macbook pro (early 2011).
I downloaded the ISO, burned it to a DVD and the boot freezes in "Done6", as shown in the photo.

[image]

SergeySlice, 2021-04-20:
I didn't test CD boot several years...

jief666, 2021-04-22:
Please try this : 
[Clover-5133-X64.iso.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6359740/Clover-5133-X64.iso.zip) and tell what you see on the screen.

Savio-Brazil, 2021-04-22:
Thank you very very much, jief666! I'll try and post the result here.

Savio-Brazil, 2021-04-26:
Hi Jief666. I just tested it. He went through "Done6" but then clean the screen and stay that way (with the screen completely black).

jief666, 2021-04-26:
What is the latest version that worked ?

Savio-Brazil, 2021-04-26:
I only tested from r5070 to here and none worked.
I did not test none before r5070.

jief666, 2021-05-05:
If Clover never ever worked, it'll be hard to debug without the hardware I'm afraid.
Anyway, knowing that's it's made for Hackintosh, have you considered trying refind (http://www.rodsbooks.com/refind/) instead. It seems more what you need...

jief666, 2021-05-20:
Closing because it looks like there is nothing we can do, but feel free to continue using the conversation thread.

closed 2021-05-20 by jief666

CE1CECL, 2021-08-21:
same issue here on vmware workstation 16.1.2 pro build 17966106

azazar, 2022-01-15:
Same for VMware® Workstation 15 Player 15.5.7 build-17171714

jiessiezhang, 2026-01-28:
VMware® Workstation 15 Pro using image Clover-5167-X64.iso

[image]