CloverHackyColor/CloverBootloader #388: Freezing right after selecting boot entry

RBFraphael, 2021-04-27:
Hi!

I'm running Hackintosh 10.15.7 with Clover 5114 for a long time (about July/2020)... Now, I'm trying to upgrade Clover version to latest (5133), but, after upgrading the EFI folder, any option on choose screen I select causes my computer to freeze... I've also tried to enable verbose, but no text output is shown.

Downgrading to 5114 again is working fine.

SergeySlice, 2021-04-28:
When you reach GUI press F2 and upload here the file /EFI/CLOVER/misc/preboot.log

RBFraphael, 2021-04-28:
Here is the working 5114 log:
[preboot-5114.log](https://github.com/CloverHackyColor/CloverBootloader/files/6393549/preboot-5114.log)

And the buggy 5133 log:
[preboot-5133.log](https://github.com/CloverHackyColor/CloverBootloader/files/6393550/preboot-5133.log)

Thanks!

SergeySlice, 2021-04-29:
With 5114 you had OcQuirks as separate file.
With 5133 you have to rewrite these quirks into config.plist. See config-sample.plist.

SergeySlice, 2021-05-15:
Seems resolved

closed 2021-05-15 by SergeySlice