CloverHackyColor/CloverBootloader #389: XMPDetection no longer accepts 'Yes/No'

laxamar, 2021-04-30:
Not critical at all, but the previous builds allowed XMPDetection to be string 'Yes' or 'No'.

We now get:
`Warning: Expect a boolean, an int8 or a string starting with 'n' or '-' for '/Boot/XMPDetection:95'`

Manually fixed config.plist

(Also /`Quirks/ProviderConsoleGopEnable:284` Is now unknown. By design?)

O1iveira, 2021-05-01:
I have the same issue.

littleOtterBOT, 2021-05-04:
[image]
Maybe they just want to change a string input to bootlean for much better management here

SergeySlice, 2021-05-04:
As far as I remember it was always string "-1", "0", "1", "2". It never was "YES".

closed 2021-05-04 by SergeySlice

laxamar, 2021-05-04:
It has always been Yes/No as far as I remember:

https://sourceforge.net/p/cloverefiboot/wiki/Boot/#xmpdetection