CloverHackyColor/CloverBootloader #390: Boot Error on 5134, 5133 is working great.

kcj83, 2021-05-01:
boot error attached for 5134, reverting back to 5133 for the time being.
[image]

yeucodonvn, 2021-05-03:
same my issue. macos 11.1

SergeySlice, 2021-05-06:
Can you provide any information to localise the bug?

kcj83, 2021-05-06:
ill do the best i can to provide any/all info i can, but it is literally crashing on boot right after splash screen/preboot selection screen.  what other type of information do you require?

SergeySlice, 2021-05-07:
Test this version
[CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6442949/CLOVERX64.efi.zip)

SergeySlice, 2021-05-15:
Is the issue valid for 5135?

SergeySlice, 2021-05-19:
abandoned - means resolved

closed 2021-05-19 by SergeySlice