CloverHackyColor/CloverBootloader #391: [OUTDATED]: please include two missing SIP values in Clover boot GUI

LeeBinder, 2021-05-01:
Right now still as of r5133 (5134 not tested yet):
```
Allow Untrusted Kexts
Allow Unrestricted FS
Allow Task For PID
Allow Kernel Debugger
Allow Apple Internal
Allow Unrestricted DTrace
Allow Unrestricted NVRAM
Allow Device Configuration
Allow Any Recovery OS
Allow Unapproved Kexts
```
Missing even though known since at least 2021-02-18 (corresponding hex value after each #):
```
# since Mojave:
CSR_ALLOW_EXECUTABLE_POLICY_OVERRIDE		# 0x0400
# since Big Sur:
CSR_ALLOW_UNAUTHENTICATED_ROOT       	 	# 0x0800
```
Source: https://bbs.archlinux.org/viewtopic.php?pid=1956991#p1956991

It would be awesome if you include the missing values so SIP can again be tweaked fully via Clover GUI as per user requirements.

Also, for reference: [ink-splatters CsrDecode branch](https://github.com/ink-splatters/CsrDecode) (branched from corpnewt), nicely generating hex values in both "flavors" for both, Clover and OC:

[image]

[image]

[image]

.. and hey, I find Clover GUI 100 x more powerful than OpenCanopy. I think they're going an inefficient route, disregarding the fact that Apple is Apple, and hackintosh is hackintosh. Just my two cents..

SergeySlice, 2021-05-04:
Accepted

SergeySlice, 2021-05-04:
Done in commit 9c6e5ce

closed 2021-05-04 by SergeySlice

LeeBinder, 2021-05-04:
Very nice, that was fast - thank you Sergey!

[image]