CloverHackyColor/CloverBootloader #393: 5134 problem

mjs520, 2021-05-04:
[image]

littleOtterBOT, 2021-05-04:
Use the CloverConfigPlistVaildiator and then check it first, updating Clover 5134 without checking the config.plist is a pretty bad idea tho

mjs520, 2021-05-04:
ok，thanks

FREYDJ, 2021-05-04:
Black screen on boot

mjs520, 2021-05-04:
> Use the CloverConfigPlistVaildiator and then check it first, updating Clover 5134 without checking the config.plist is a pretty bad idea tho

how to Use the CloverConfigPlistVaildiator

FREYDJ, 2021-05-04:
> > Используйте CloverConfigPlistVaildiator, а затем сначала проверьте его, обновление Clover 5134 без проверки config.plist - довольно плохая идея, хотя
> 
> как использовать CloverConfigPlistVaildiator
Закинь его в terminal + свой config. Далее жми enter

mjs520, 2021-05-04:
> > > Используйте CloverConfigPlistVaildiator, а затем сначала проверьте его, обновление Clover 5134 без проверки config.plist - довольно плохая идея, хотя
> > 
> > 
> > как использовать CloverConfigPlistVaildiator
> > Закинь его в terminal + свой config. Далее жми enter

can you speak English?thanks

FREYDJ, 2021-05-04:
> > > > Используйте CloverConfigPlistVaildiator, а затем проверьте его, обновление Clover 5134 без проверки config.plist - довольно плохая идея, хотя
> > > 
> > > 
> > > как использовать CloverConfigPlistVaildiator 
> > > Закинь его в терминал + свой config. Далее жми введите
> 
> ты говоришь по английски? спасибо

My English is very bad(Google translate)😂 my language is Russian

FREYDJ, 2021-05-04:
> > > > Используйте CloverConfigPlistVaildiator, а затем проверьте его, обновление Clover 5134 без проверки config.plist - довольно плохая идея, хотя
> > > 
> > > 
> > > как использовать CloverConfigPlistVaildiator 
> > > Закинь его в терминал + свой config. Далее жми введите
> 
> ты говоришь по английски? спасибо

drags CloverConfigPlistVaildiator into the terminal then your config. then press enter. it will show you what to fix in the config ... but my laptop does not start, stupidly black screen

kushwavez, 2021-05-04:
Just fix the issues mentioned on the screen.

XMPDetection = Boolean
"Unknown key, Skipped" means there are no such key in this release, ignored, they should be deleted (except DeviceProperties and Custom GUI elements)
Last one is obvious, put it into a boolean instead of <string>Yes</string>

As for the black screen issue, try and compare it to config-sample.plist from Windows for example.

FREYDJ, 2021-05-04:
> Просто исправьте проблемы, указанные на экране.
> 
> XMPDetection = Boolean
> «Неизвестный ключ, пропущен» означает, что в этом выпуске нет такого ключа, игнорируется, их следует удалить (кроме DeviceProperties и пользовательских элементов графического интерфейса).
> Последний очевиден, поместите его в логическое значение вместо «Да»
> 
> Что касается проблемы с черным экраном, попробуйте сравнить ее, например, с config-sample.plist из Windows.

i was building in mac os x from a clean sample

z3DD3r, 2021-05-04:
> Black screen on boot

How did you build Clover? Using Xcode? If so then try to build it with GCC. Type `sh buildme` in terminal an select `build all`. And don't forget to update your config as @kushwavez said.

SergeySlice, 2021-05-04:
It is not a Clover issue. It is the issue with your config.plist.

closed 2021-05-04 by SergeySlice