CloverHackyColor/CloverBootloader #395: Issue (closed) #293

scerazy, 2021-05-09:
https://github.com/CloverHackyColor/CloverBootloader/issues/293

It should not be closed, because it is only partially solved

sebus

kushwavez, 2021-05-09:
What I do: 
- Download BaseSystem.dmg with this python script (`./macrecovery.py -b Mac-E43C1C25D4880AD6 -m 00000000000000000 download`): https://github.com/acidanthera/OpenCorePkg/blob/master/Utilities/macrecovery/macrecovery.py
- Format USB to HFS+ with ex. name "Untitled"
- Write/Restore BaseSystem.dmg to the USB Untitled partition
- Mount EFI and install Clover, etc (no need if you already have it on your system)
- Restart, select the Installer from the Clover GUI (or if you have EFI on USB, boot from it)
Note that it's for Big Sur Recovery, not sure if this works for older OSs.

You can also do that from Win : https://www.insanelymac.com/forum/topic/346703-guide-creating-clover-macos-big-sur-installer-usb-on-windows/

scerazy, 2021-05-22:
I am not interested in doing it this way. Ofcours eit works. But I like just copy of the UUID named folder to USB & in Refind it works as is, seems that Clover has hard-coded search for com.apple.recovery.boot folder, that really could be corrected

scerazy, 2021-05-23:
Sure, close it, because... you do not like it

jief666, 2021-05-24:
I'm not sure what you'd like to do.
I'm against improving, but it has to be of some use.
Any particular reason you want to copy that UUID folder instead of leaving it next to the bootable recovery partition ?

kushwavez, 2021-05-24:
I think he want to boot the BaseSystem.dmg the OC way, so just simply put BaseSystem.dmg inside a FAT32 partitioned USB drive, then select the boot entry from OC UI. 

It is easy and very quick method to for example create an Internet Recovery Installer from whatever OS you have, if you have no macOS installed. 

Clover, from v5129(?) can boot Recovery partition (so can boot the BaseSystem.dmg), but can't boot BaseSystem.dmg from USB drives.

jief666, 2021-05-25:
Should Clover propose a boot entry for any BaseSystem.dmg it can find ?
Or improve the custom entries so one can be made ?

kushwavez, 2021-05-25:
> Should Clover propose a boot entry for any BaseSystem.dmg it can find ?

Yes, that is a good idea.

jief666, 2021-05-26:
If I take a baseSystem.dmg from a recovery, I put it **alone** on a usb key, does it make something bootable ? It won't look for other component somewhere else ?

scerazy, 2021-05-29:
So could we get away from hard-coded com.apple.recovery.boot" ?

jief666, 2021-05-29:
I don’t see how we can get away with that. Searching for every dmg in all volume is too long. But we can allow searching the hard coded folder on hfs and even fat32 volumes.

scerazy, 2021-05-30:
So how does Refind do it (it displays the boot option instantly)

kushwavez, 2021-05-30:
> But we can allow searching the hard coded folder on hfs and even fat32 volumes.

This is the way

scerazy, 2021-05-30:
Hard-coded is definitely not the way to go! 
I explained in first post exactly what I would expect
https://github.com/CloverHackyColor/CloverBootloader/issues/293
It is way easier to just copy the existing UUID folder from local Recovery partition then needlessly create recovery USB by download
Refind does it fine (presents boot from Recovery in its menu), without need for hard-coded name of the folder

jief666, 2021-05-31:
> Hard-coded is definitely not the way to go!
Are you saying that "com.apple.recovery.boot" is NOT hard coded in refind ?

kushwavez, 2021-05-31:
I really think that it is hard-coded in refind too and searching for a dir called com.apple.recovery.boot, then BaseSystem.dmg on every drive. I think we misunderstood each other. 

In your other issue you wrote that `showtools apple_recovery` setting in Refind config can enable the recovery USB, and can find IF the BaseSystem.dmg is in that com.apple dir. So, that's a hard-coded path. 

Not hard-coding the path would result in a significantly slow entry creation process in my opinion.

scerazy, 2021-06-10:
I never had to use com.apple.recovery.boot while using Refind. I just copy UUID named folder from Recovery partition & it is being used

scerazy, 2021-07-14:
Anybody any more?

jief666, 2021-07-14:
I'll try something soon.
Also, you never answered the question I asked months ago...

scerazy, 2021-10-02:
Did you mean:

Any particular reason you want to copy that UUID folder instead of leaving it next to the bootable recovery partition ?

Then this question makes no sense, the UUID folder is a copy from existing HD Recovery partition
Put it on USB & for Refind that IS the Recovery USB bootable disk

jief666, 2021-10-03:
> Then this question makes no sense
If I asked, it is because it makes sense. Please give me at least that credit.
So I guess what you mean is that you didn't understand the question.

I was curious about why you do need a standalone recovery partition on a USB stick.

> Put it on USB & for Refind that IS the Recovery USB bootable disk
Yes, I know, but if don't have a main partition to recover, what would you do with a recovery partition if you have nothing to recover ?

scerazy, 2021-10-04:
Because Recovery is the most minimal MacOS boot (like Windows PE), and it does not relay on internal drive existing and be in usable state
Booting from USB I can apply any image I want to the internal disk with asr
Or test anything else (which app might run in that minimal environment?), or read data of the existing disk if ie corrupted

I am honestly surprised at such question

jief666, 2021-10-04:
Use installer instead of recovery...

> 1) Then this question makes no sense
> 2) I am honestly surprised at such question

Maybe you should try just answering nicely and avoid this kind of comments that :
 - don't bring anything constructive to the discussion
 - won't make me want to work for you for free

scerazy, 2021-10-04:
You are not working for me, not paid or free. You do work (on this) because you want to. And I only provide feedback which can make product better

But you are welcome to ignore my input

jief666, 2021-10-04:
> You are not working for me

Making a feature that only **you** want, _is_ working for you. I would have only done it to be nice.

> You do work (on this) because you want to

I work on Clover because I want to. I would have worked on your feature because **you** want it.

> But you are welcome to ignore my input

Thank you very much for your permission.

closed 2021-10-04 by jief666