CloverHackyColor/CloverBootloader #396: update Clover 5134 and black screen

loulich, 2021-05-09:
Hello, i've updated clover on last version 5134 and on reboot black screen..
I read the previous issues and i've use CloverConfigPlistValidator and it's ok well done.

I've no information with -v verbose because crash immediately
If you have a solution..
Thanks

LaCruz75, 2021-05-10:
Same here with 5134.
-v not working. Just can choose boot selection, immediately black screen. No logs.

Going back to 5133 all fine.

ryanm352, 2021-05-10:
> Hi, take a look and read here: https://applelife.ru/threads/clover.42089/page-1386#post-940369 👉/CLOVERX64.efi/
> 
> Helped for my old car; GA-P55-US3L Lynnfield Intel® Core™ i7-870 Cape Verde PRO /Radeon HD 7750/
> 11.2.3 (20D91)+Catalina+Mojave  Apple ROM:Apple ROM Version.Board-ID : Mac-F2238BAE
> ⌘ Powered by Clover revision: 5134 🙌

Just tried that CLOVERX64.efi, same black screen. Also tried one from a few issues down in github, same thing. Had to revert to 5133. Running 11.2.3

loulich, 2021-05-11:
Hi, with this file CLOVERX64.efi it's ok for me !
https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2757547

ryanm352, 2021-05-11:
> Hi, with this file CLOVERX64.efi it's ok for me !
> https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2757547

This worked along with the instructions! thanks!!

SergeySlice, 2021-05-15:
Is the issue still valid?

loulich, 2021-05-15:
Hi, for me it's ok, in first time i've modify my plist and remove all DSDT options.
No black screen but lag, bug and the update to 11.3.1 not finishing.
After i've changing the file CLOVERX64.efi and replace the plist with the old (the first with dsdt options) and all is ok.
Boot ok and update finishing !

closed 2021-05-19 by SergeySlice