CloverHackyColor/CloverBootloader #397: High Definition Audio Controller Code 10

meakcey, 2021-05-10:
I have a PCIe-m.2 SSD on PCI2.0 16x and graphic card on PCI2.0 4x
Dual boot with Windows 10 and Ubuntu
Succesffully could boot both OS but audio driver does not work on both.
It is not related with Windows driver nor hardware. If I boot regular way with another disk on SATA, same driver with same version working.
I have deleted AudioDxe from Clover UEFI but didn't work. If I keep this .efi file, some white noise coming out constantly.
All combinations uninstall, reinstall, disable, enable, stop windows update, reinstall updated drivers but no luck.
Something preventing driver to work on properly or bus getting stuck, I don't know.
Any help appreciated.

SergeySlice, 2021-05-15:
Try to set in config.plist
~~~~
	<key>Devices</key>
	<dict>
		<key>Audio</key>
		<dict>
			<key>ResetHDA</key>
			<true/>
		</dict>

~~~~

meakcey, 2021-05-17:
Current section is like below

<key>Audio</key>
		<dict>
			<key>#Inject</key>
			<string>12</string>
			<key>AFGLowPowerState</key>
			<true/>
			<key>ResetHDA</key>
			<true/>
		</dict>

Should I change completely as you requested?

meakcey, 2021-05-17:
Did not solve.
I also get back the AudioDxe.efi driver from "off" section
Uninstalled and installed HDA Controller 
No luck.

Polaris35, 2022-07-14:
did you find a solution to the problem? I have the same problem

Polaris35, 2022-07-17:
@meakcey hi, did you find a solution to the problem?