CloverHackyColor/CloverBootloader #407: Clover can't boot x99

etajv100, 2021-05-19:
Hi, i'm having trouble trying to boot big sur on a x99 machine,

cpu i7 5820k
x99s gaming 7 msi (ms 7885)
gtx 680/gtx 770 (i have both only one installed)

any help would be appreciated.

SergeySlice, 2021-05-19:
May be your CPU is unsupported? What is CPU signature?

Guys,
always provide debug.log if open new issue.

etajv100, 2021-05-19:
How do i check cpu signature on mac, and where do i find debug.log please?

jief666, 2021-05-20:
So far, it doesn't look like a Clover issue, just a config problem. For that, forums might help, as we don't have enough time to repeat over and over again the same things (like how to activate debug, where is the file, two to configure for a specific hardware, etc.).

Closing the issue, but we can still use this conversation thread. When you'll have a debug log, post it here.

closed 2021-05-20 by jief666