CloverHackyColor/CloverBootloader #408: Unable to boot r5135 - config.plist issue

SH-M, 2021-05-20:
Hi,

I have been running macOS 11.2.3 and Clover r5126. Today I tried to update to r5135 but on booting, clover reported that my config.plist had errors and to validate it with CCPV. 

I have attached the output from it.
[image]

Also attached is my EFI.
[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6515459/EFI.zip)

Specs:
Lenovo y700-15ISK
i7-6700HQ
HD530 + GTX960m(disabled), 1920x1080

Any help would be appreciated, thanks.

jief666, 2021-05-20:
Delete unknown keys, and delete
```
			<dict>
				<key>Signature</key>
				<string>#MCFG</string>
			</dict>
```

I'm closing the issue, but we can continue to write here.

SH-M, 2021-05-20:
@jief666 
Hi,
Thanks for helping out. 

I have removed the lines you mentioned, and that has made CCPV happy. Now I am getting past to the clover boot screen but the MacOS boot option has disappeared and even when I check the hidden ones I don't see the right one.

Here's the updated config.plist
[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6517750/config.plist.zip)

And here is the preboot log
[preboot.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6517773/preboot.log.zip)

jief666, 2021-05-21:
Ok. Send a log of 5126 so I can compare.

SH-M, 2021-05-21:
@jief666 
Here's the log from r5126
[preboot r5126.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/6522062/preboot.r5126.log.zip)

SH-M, 2021-05-25:
Hi,
I have fixed the issue. All I had to do was use the preboot volume.

jief666, 2021-05-25:
Good. Thanks for telling me.