CloverHackyColor/CloverBootloader #409: Update to 5135 But Can't Boot

HandaTakuya, 2021-05-21:
I update Clover version 5133 to 5135 but Can't boot to Desktop

[image]

I check config.plist with CloverConfigPlistValidator is say "Your plist looks so wonderful. Well done!"

[image]

mjs520, 2021-05-21:
卡内存了吧

jief666, 2021-05-24:
Can you boot with this efi file : https://github.com/jief666/CloverCommits/blob/master/C-20210411074317-fd3b09c-5133.efi ?
If not, send a the log file.

HandaTakuya, 2021-06-05:
Yes, I can boot efi from 5133

jief666, 2021-06-06:
And this one : https://github.com/jief666/CloverCommits/blob/master/C-20210429211313-fb86ee1-5134.efi

SergeySlice, 2021-06-09:
> Yes, I can boot efi from 5133

IIs there the same EFI folder with the same files but Clover?
What EFI/CLOVER/drivers/UEFI installed?

HandaTakuya, 2021-06-10:
> > Yes, I can boot efi from 5133
> 
> IIs there the same EFI folder with the same files but Clover?
> What EFI/CLOVER/drivers/UEFI installed?

in EFI/CLOVER/drivers/UEFI
[image]

now I new create config from 5136 simple-config and success to boot in Clover 5136 But... my Laptop is Black Screen. Can login to Desktop and connect extend Display to use Hackintosh. Then another 5-10 minutes My Laptop Display is Back to work again.

HandaTakuya, 2021-06-10:
> in EFI/CLOVER/drivers/UEFI
> [image]
>
> now I new create config from 5136 simple-config and success to boot in Clover 5136 But... my Laptop is Black Screen. Can login to Desktop and connect extend [...]

After update 5136 my Laptop is Black Screen. Can login to Desktop and connect extend Display to use Hackintosh

[image]

Then another 5-10 minutes My Laptop Display is Back to work again.

[image]

HandaTakuya, 2021-06-10:
Update : I can FIXED ! 

Thank for Help ! ><

closed 2021-06-10 by HandaTakuya

jief666, 2021-06-17:
Could you explain what was the problem and how you solved it. It might help others...

HandaTakuya, 2021-06-17:
> Could you explain what was the problem and how you solved it. It might help others...

create new config from 5136 simple-config and remove other key