CloverHackyColor/CloverBootloader #411: Clover r5135 broke booting - Coming from r5120

gateway, 2021-05-23:
Ugg.. So I was planning on upgrading to Big Sur from Catalina (latest version build) and I went to use clover configurator and updated the clover build though the system. This installed 5135 and now when I boot it goes for about 4 seconds and then reboots and this continues over and over.

Trying the recovery partition also does the same.

I managed to take a screenshot but it was so freaking fast..

Bottom line says in memory panic stackshot succeeded ** bytes traced 15296 **

I have attached my config.plist (removed serial etc..)

I have tried to create a bootable catalina usb drive by installing catalina on usb and installed clover builds. I have tried a few but when I boot off of that I also get the same type of result.

Any thoughts? My data is still there just freaking need to figure out how to get this to boot now :/

[image]

```<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>ACPI</key>
	<dict>
		<key>SSDT</key>
		<dict>
			<key>DropOem</key>
			<false/>
			<key>Generate</key>
			<dict>
				<key>CStates</key>
				<false/>
				<key>PStates</key>
				<false/>
				<key>PluginType</key>
				<true/>
			</dict>
		</dict>
		<key>AutoMerge</key>
		<true/>
		<key>DSDT</key>
		<dict>
			<key>Fixes</key>
			<dict>
				<key>FixIDE</key>
				<false/>
				<key>AddHDMI</key>
				<false/>
				<key>FixIntelGfx</key>
				<false/>
				<key>FixACST</key>
				<false/>
				<key>FakeLPC</key>
				<false/>
				<key>FixFirewire</key>
				<false/>
				<key>FixShutdown</key>
				<false/>
				<key>FixDisplay</key>
				<false/>
				<key>AddIMEI</key>
				<false/>
				<key>FixHPET</key>
				<false/>
				<key>AddMCHC</key>
				<false/>
				<key>FixAirport</key>
				<false/>
				<key>FixDarwin7</key>
				<false/>
				<key>FixRTC</key>
				<false/>
[... 1434 lines cut ...]
59:545  0:007  OCAK: Requesting SymbolTable for com.insanelymac.RealtekRTL8111 - 618
59:562  0:017  OC: Prelinked injection Kexts\Other\RealtekRTL8111.kext () - Success
59:569  0:007  OCAK: org.vulgo.NoVPAJpeg got KXLD 0 0
59:576  0:007  OCAK: Requesting __LINKEDIT for org.vulgo.NoVPAJpeg - 7F1E53E8 at 6A5E000
59:584  0:007  OCAK: Requesting SymbolTable for org.vulgo.NoVPAJpeg - 314
59:597  0:013  OC: Prelinked injection Kexts\Other\NoVPAJpeg.kext () - Success
59:606  0:009  OCAK: com.rehabman.driver.USBInjectAll got KXLD 0 0
59:614  0:007  OCAK: Requesting __LINKEDIT for com.rehabman.driver.USBInjectAll - 7F1ED398 at 6A66000
59:621  0:007  OCAK: Requesting SymbolTable for com.rehabman.driver.USBInjectAll - 351
59:637  0:015  OC: Prelinked injection Kexts\Other\USBInjectAll.kext () - Success
59:644  0:007  OC: Prelink size 111611904 kext offset 108683264 reserved 2928640
59:665  0:021  OC: Kernel patch [0] AppleAHCIPort (External icons patch)
59:676  0:010  OCAK: Failed to pk find AppleAHCIPort - Not Found
59:683  0:007  OC: Kernel patcher result 0 for AppleAHCIPort (AppleAHCIPort (External icons patch)) - Not Found
59:693  0:009  OCAK: com.apple.driver.AppleIntelCPUPowerManagement got KXLD 0 0
59:701  0:008  OCAK: com.apple.driver.AppleRTC got KXLD 0 0
59:709  0:007  Replace BE58000000/FFFEFFFFFF(BE58000000) by BEFFFF0000 at ofs:1C89
59:900  0:191  Replace BE58000000/FFFEFFFFFF(BE59000000) by BEFFFF0000 at ofs:1C99
60:090  0:189  Replace BE58000000/FFFEFFFFFF(BE58000000) by BEFFFF0000 at ofs:1F48
60:278  0:188  Replace BE58000000/FFFEFFFFFF(BE59000000) by BEFFFF0000 at ofs:1F59
60:465  0:187  OCAK: DisableRtcChecksum replace count - 4
60:473  0:007  OCAK: Patch success com.apple.driver.AppleRTC DisableRtcChecksum
60:491  0:018  OC: Prelinked status - Success
60:521  0:029  ++++++++++++++++++++++++++++++++
```

gateway, 2021-05-23:
Did I miss something that needs to change in the config or why am I getting this over and over :/

Khaled-Wolf, 2021-05-23:
I have the same issue, same previous clover 5120 and same version I am upgrading to 5135, also same error

stevezhengshiqi, 2021-05-24:
Check [config-sample](https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist), you need `Quirk` section.

SergeySlice, 2021-05-26:
Don't use VirtualSMC and FakeSMC at the same time. They are in conflict.
My advice is to remove VirtualSMC.* (.efi and .kext)

gateway, 2021-05-27:
> Check [config-sample](https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist), you need `Quirks` section.

it seems empty.. <key>Quirks</key> .. unfortunately this is pretty new to me.  I wish we had some good documentation on migration from older 51xx to latest 5135.  Ill try to load the plist file on my macbook in clover config and see if I can add quirks, however its not clear what you add.. :/

gateway, 2021-05-27:
> I have the same issue, same previous clover 5120 and same version I am upgrading to 5135, also same error

did you find a way to fix this?

gateway, 2021-05-28:
So I'm finally backup and thanks for anyone that pointed me in the right direction..

For those who may be facing something similar I had to add the following quirks to my config.plist based upon my CPU/Motherboard.

Link to the core i7-8700 cpu [url=https://dortania.github.io/OpenCore-Install-Guide/config.plist/coffee-lake.html](https://dortania.github.io/OpenCore-Install-Guide/config.plist/coffee-lake.html)

If you have a different cpu on the left hand side is a link to the various ones so select that and go to your quirks area.

My specific changes to my plist are the following.

```
    <key>Quirks</key>
    <dict>
        <key>DevirtualiseMmio</key>
        <true/>
        <key>EnableWriteUnprotector</key>
        <false/>
        <key>ProtectUefiServices</key>
        <true/>
        <key>RebuildAppleMemoryMap</key>
        <true/>
        <key>SyncRuntimePermissions</key>
        <true/>
    </dict>
```

So thanks for anyone that posted info to help.. hopefully this will help someone in the future.

SergeySlice, 2021-05-31:
Anyway issue tracker is not a school for hackintosh.
Open new issue with a description of Clover mistake if any.

closed 2021-05-31 by SergeySlice

gateway, 2021-05-31:
One thing to note, while not a code issue this is a documentation issue.  With the move to add opencore and its required quirks based upon cpu/motherboard I think the main readme should reflect a link to the opencore cpu section and its recommended quirks settings otherwise you will continue to have questions asked over and over..