CloverHackyColor/CloverBootloader #415: Could you package Itlwm into an EFI driver?

1457384613gh, 2021-05-28:
Could you package them into an EFI driver ?
As is known to us, Macintosh can connect to the WiFi when user choose to boot a choice, which is used for booting from wireless network.
If [Itlwm](https://github.com/OpenIntelWireless/itlwm) could be packaged into an EFI driver, Hackintosh would be superior and more similar to Macintosh.
[Itlwm](https://github.com/OpenIntelWireless/itlwm)
https://github.com/OpenIntelWireless/itlwm/issues/608

SergeySlice, 2021-05-31:
We are not including outside software into our package. We include only those that can be compiled from our sources.

closed 2021-05-31 by SergeySlice

1457384613gh, 2021-06-01:
> We are not including outside software into our package. We include only those that can be compiled from our sources.

Thanks for your reply and how earnest you are !