CloverHackyColor/CloverBootloader #416: Black screen on wake-from-sleep since DSDT / SlpSmiAtWake was dropped -> substitute or restoration needed

LeeBinder, 2021-06-05:
With recent Clover, DSDT / [SlpSmiAtWake](https://sourceforge.net/p/cloverefiboot/wiki/ACPI/#dsdt-slpsmiatwake) was dropped. For my Asus VivoBook, that key is needed to prevent an intermittent (not always) black screen on wake-from-sleep. I just encountered such blank screen and realized this key not present in my config.plist anymore.

Why did you drop this key - what bad did it do? And is there a substitute, like a boot param to directly add `SLP_SMI_EN=0` at every wake, or even a new Quirk?

SergeySlice, 2021-06-06:
I am sorry but how old Clover is still contain this key? I don't remember it.
An I recommend to you forget it.

closed 2021-06-06 by SergeySlice

LeeBinder, 2021-06-06:
wait wait @SergeySlice , it doesn't work that way.. just by looking away and denying a fact. Like I said, the key is required for smooth operation; totally regardless of how "old" it is. I bet there are "older" keys still active and being used by others. Why would "age" matter for any key or parameter? That doesn't make any sense/ qualify as an argument. And looking around the web there are others who use this key, too.

So please, would you (or another dev) please restore this key, or otherwise offer a working substitute? Those who don't need that key, simply don't activate it, and those like Asus VivoBook S15 users who need it, should have the possibility to  turn it on. Simple as that - no problem at all.

Thank you.

SergeySlice, 2021-06-06:
Did you show me this key in any sources?

LeeBinder, 2021-06-06:
If I could I would but can't because I'm not a dev and don't know the sources. All I observed that it got dropped recently - with 5134 maybe? What if you search the pre-5134 source code (5133 and maybe prior) for `SlpSmiAtWake` and `SLP_SMI_EN`?

SergeySlice, 2021-06-06:
How did you know the patch ever exists? In what version?

LeeBinder, 2021-06-06:
I can't tell you but can only go by evidence, which is: with that key activated, I hadn't had a black screen on wake for ages, until just recently. Just from that, I'd consider it highly probable that `SlpSmiAtWake` functionality was still present in at least 5132, to be on the safer side with a specific version number.

SergeySlice, 2021-06-06:
It is absent in 5103 as well.
I am not sure it ever work up to 4400

LeeBinder, 2021-06-06:
highly strange, that beats me, and thanks for checking - don't know what to say...

`SlpSmiAtWake` is one of three keys I activated to combat black-screen-on wake as per [this posting](https://www.hackintosh-forum.de/forum/thread/36237-sierra-high-sierra-black-screen-nach-wake-wahrscheinliche-l%C3%B6sung/) - the other two are `HaltEnabler` and `SuspendOverride` which I still have activated.

I need to admit that this time the KP content looks different from the times in the past before:
```
mp_kdp_enter() timed-out on cpu 6, NMI-ing
mp_kdp_enter() NMI pending on cpus: 0 1 2 4
mp_kdp_enter() timed-out during locked wait after NMI;expected 6 acks but received 2 after 6720091 loops in 900000000 ticks
panic(cpu 6 caller 0xffffff800c8e4b56): "mp_rv_action() exit: 4 of 5 responses, start: 0x25ecdc5c, cur: 0x23815d35a"@/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-4903.278.68/osfmk/i386/mp.c:918
Backtrace (CPU 6), Frame : Return Address
0xffffff81168fcbb0 : 0xffffff800c7ad5cd 
0xffffff81168fcc00 : 0xffffff800c8e9245 
0xffffff81168fcc40 : 0xffffff800c8da97a 
0xffffff81168fccb0 : 0xffffff800c75a9d0 
0xffffff81168fccd0 : 0xffffff800c7acfe7 
0xffffff81168fcdf0 : 0xffffff800c7ace33 
0xffffff81168fce60 : 0xffffff800c8e4b56 
0xffffff81168fcea0 : 0xffffff800c8e3eb3 
0xffffff81168fcf10 : 0xffffff800c8e3b93 
0xffffff81168fcf50 : 0xffffff800c8e2a93 
0xffffff81168fcf70 : 0xffffff800c8da0cb 
0xffffff81168fcfd0 : 0xffffff800c75ab7d 
0xffffff912463bf30 : 0xffffff800c7cf207 
0xffffff912463bf80 : 0xffffff800c7cf3f0 
0xffffff912463bfa0 : 0xffffff800c75a0ce 

BSD process name corresponding to current thread: kernel_task
Boot args: #keepsyms=1 #kext-dev-mode=1 -v brcmfx-country=DE alcid=13 

Mac OS version:
18G9216

Kernel version:
Darwin Kernel Version 18.7.0: Mon May  3 20:41:19 PDT 2021; root:xnu-4903.278.68~1/RELEASE_X86_64
Kernel UUID: 49D0FC60-2676-3913-8901-2E4C5603365D
Kernel slide:     0x000000000c400000
Kernel text base: 0xffffff800c600000
__HIB  text base: 0xffffff800c500000
System model name: MacBookPro15,4 (Mac-53FDB3D8DB8CA971)

System uptime in nanoseconds: 4887421093259
```

LeeBinder, 2021-06-06:
just checked via `pmset -g | grep hibernatemode`: it's set to 0 as supposed. To me this KP on wake-from-sleep does not ring a familiar bell..