CloverHackyColor/CloverBootloader #417: Clover v5136. Usb keyboard does not work, but usb mouse become to.

dioxine, 2021-06-11:
Hello.
I've got kind of strange behaviour after updating from v5136 from v5135 version of Clover. My usb mouse never worked in Clover GUI, but after updating, my mouse now works finely, but there is no reaction from usb keyboard key presses. 
I use legacy version of Clover cause my old motherboard does not have UEFI, only old BIOS from AMI.
Tried to put  `usbkbdxe.efi` and  `usbmousedxe.efi` in `drivers\BIOS` folder, but there was no effect. Anyway i've never used them before and keyboard was functional.
```
Motherboard: ASUS X58 Sabertooth with Intel X58 chipset
CPU: Intel Xeon X5650
GPU: AMD Radeon RX 580 4gb
Memory: 24gbs, triple channel
```

SergeySlice, 2021-06-18:
Make please debug.log by setting
config.plist->Boot->Debug=true
After boot you can find new file in /EFI/CLOVER/misc
attach it here.

juanpc2018, 2021-06-19:
Same here:
5137.iso
Genuine Apple Mac Mini 2014
Microsoft Wireless USB All-In-One Keyboard + Trackpad.

juanpc2018, 2021-06-19:
i cannot change config.plist file, 
because i´m Not the owner 502
i tested all i could in Kubuntu 20.10
sudo apt-get install hfsprogs hfsplus

$ sudo mount -o rw,remount /dev/sdc2  /media/me/Clover-5137-X64
$ sudo chown me /media/me/Clover-5137-X64/EFI/CLOVER/config.plist
chown: changing ownership of '/media/me/Clover-5137-X64/EFI/CLOVER/config.plist': Read-only file system
$ sudo chmod +w /media/me/Clover-5137-X64/EFI/CLOVER/config.plist
chmod: changing permissions of '/media/me/Clover-5137-X64/EFI/CLOVER/config.plist': Read-only file system
$ ls -l /media/me/Clover-5137-X64/EFI/CLOVER/config.plist
-rwxr-xr-x 1 502 dialout 27862 Jun 18 12:06 /media/me/Clover-5137-X64/EFI/CLOVER/config.plist

dioxine, 2021-06-21:
Hey! Now in v5137 usb keyboard works again and again there is no usb mouse :) i don't really need mouse. Here are two logs attached, from v5136 and v5137.
[v5137-2021-06-21_02-28_CLOVERX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/6683681/v5137-2021-06-21_02-28_CLOVERX64.EFI.log)
[v5136-2021-06-21_02-21_CLOVERX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/6683682/v5136-2021-06-21_02-21_CLOVERX64.EFI.log)

closed 2021-09-03 by SergeySlice