CloverHackyColor/CloverBootloader #419: Fail OSX Mavericks in Mac Mini 2014 & install Tips in Linux & Need Help BIOS

juanpc2018, 2021-06-19:
#1. so far "it´s impossible to run OSX Mavericks 10.9.5 in a Mac Mini 2014 using USB 2.0"
hate that S**t about Apple.
https://en.wikipedia.org/wiki/Mac_Mini#Supported_operating_systems

Latest Release of OSX Mavericks was in 2016
https://en.wikipedia.org/wiki/OS_X_Mavericks 
https://en.wikipedia.org/wiki/MacOS_version_history

i haven´t tried OSX Snow Leopard 10.6.8
but definitely will try,
at the moment of install allows option install Rosetta v1 
Rosetta v1 allows to run PowerPC G4 G5 software on iintel x86_64.
there is a lot of Abandonware in MacRepository and MacGarden.

anyway...
#2. All the tutorials i´ve seen installing CloverCD.iso in Linux are Crazy Complex Terminal Crazy.
https://www.aioboot.com/en/clover-bootloader-windows/
http://web.archive.org/web/20190330210524/https://clover-wiki.zetam.org/Installation

Easy Method Kubuntu:
sudo apt-get hfsplus
sudo apt-get apfs
sudo apt-get gnome-disk-utility
sudo apt-get fuseiso
or use Muon to install packages.
USB-Disk-Image does Not work.

Step 1.
plug a USB drive, any size, 8GB or less.
mount the Clover .iso as a standard virtual drive, i´m using Kubuntu 20.04 LTS
open the virtual drive to see the folder, then close Dolphin File Browser
open Disk utility:
A) Select & Format the Real USB Drive Fat32, 
B) Select the virtual drive, click the 3 dots up right corner, near close window.
Create image, save in Documents folder: CloverCD.iso

if you select individual partitions, you can also create images of individual partitions, 
but that is Not useful in this scenario.
create a whole single image.
 
C) Select Real USB Drive, click the 3 dots up right corner, Restore Image,
select the CloverCD.iso
DONE.
Works OK in most Boards.

Problem #2.
i need a BIOS CD image, Not UEFI.
for older Server Boards.
UEFI USB is Not detected.

SergeySlice, 2021-06-30:
What is the issue? Clover package can create bootable USB flash drive and bootable ISO is also provided.

juanpc2018, 2021-06-30:
booting from USB3.0 port:
OSX Mavericks has a forbidden logo, = Mac Mini EFI needs to be jail broken,
OSX SnowLeopard does Not have forbidden logo, has apple boot screen, but stops loading.
SnowLeopard does Not have USB3.0 drivers, or Boot capability from USB3.0

OSX boots easy from USB than Windows,
Windows is limited to USB2.0 Only.
OSX some can boot with USB3.0, Not older OSX.
 
maybe will try later using SATA port.
but... Mac Mini 2014 has SATA-III 6Gbps,
Mac Mini 2010 have SATA-II 3Gbps.
both AHCI,
maybe could work, maybe.

i have to completely disassemble the Mac Mini to change the SATA drive.
requires special Torx 6 8 screw tools. ("i have already")
https://www.ifixit.com/Teardown/Mac+mini+Late+2014+Teardown/30410

Booting from USB3.0 i don´´t see anyhope using older OSX.
OSX Sierra boots ok from USB3.0, easy.

SergeySlice, 2021-07-06:
Never give up!

closed 2021-07-06 by SergeySlice