CloverHackyColor/CloverBootloader #422: How to turn off SIP under macos12, thank you

mjs520, 2021-06-27:
How to check when patching, whether SIP is closed, and how to close SIP under clover

kushwavez, 2021-06-29:
`csrutil status`

0xFEF : Disable SIP

SergeySlice, 2021-06-30:
If you check and modify SIP UNDER CLOVER then enter Clover GUI, choose "Options" and then foollow the screenshots
[image]
[image]
[image]
Some questions remain?
This is issue tracker and not a school how to use Clover

closed 2021-06-30 by SergeySlice