CloverHackyColor/CloverBootloader #423: I cannot Run Windows from USB in Microsoft Surface Pro 2 Tablet

juanpc2018, 2021-06-27:
some people run Android emulators in Microsoft Surface Pro 2,
but i cant run Windows in Microsoft Surface Pro 2 tablet. ???

https://en.wikipedia.org/wiki/Surface_Pro_2

i can Run Kubuntu 20.10 from USB, Not Windows 8.1 from USB

the idea is to use Windows from external USB drive,
because the internal Lite-On 64GB mSATA drive is failing,
replacing the drive is a nightmare, nested deep inside.
https://www.ifixit.com/Teardown/Microsoft+Surface+Pro+2+Teardown/18604

internal batteries are also damaged, but works Ok with original AC DC adaptor.
touch screen works ok,
screen works ok,

the only problem is the very limited Bios / EFI

the tablet must enter in Bios configuration mode,
by windows setup, or Gparted bios setup,
or by pressing the Up Volume Kay while power on the tablet.

Trusted Platform must be dissabled,
Secure Boot Bios also must be Dissabled.

Boot Screen becomes Ugly RED, Not Black,
but allows to boot Kubuntu from USB2.0 using a powered USB2.0 Hub 
the Tablet has USB3.0 but i don´t have a USB3.0 hub.
ACME Hubs fail too much, specially those with many leds and switches.

anyway... Kubuntu works ok from USB,
but sometimes i need Windows.
PlayOnLinux and Wine 6.11 are getting very good, but still some software require a Real Windows.

the idea is to boot Windows8.1x64 from USB, Not internal Drive..

i made a image back up, so i can restore again later to a new mSATA or a New computer if i need.
Windows USB Recovery drive works ok.

any ideas?
worse case scenario would be moving the real windows drive to a Virtual Box inside linux.
but...
Surface Pro 2 only has 2 cores, Not a good idea to have 1-core on Windows.
Surface Pro 2 only has 4GB of Ram, Not enough to make a Virtual Machine.

also Virtualization is Not 100% the same, some external hardware does Not work, like FireWire for example.
USB works Ok.

incredible i cant run Windows on a Microsoft tablet.

jief666, 2021-06-28:
So, it's not a Clover issue, is it ?
You should install Windows (probably Windows to go) using your limited bios, just to be sure it can be done on the tablet. Maybe it can't. Maybe microsoft decided not to support that. Maybe you need to add USB driver to Windows installer.
Once Windows is up and running, if Clover can't boot it, we might have a look.

closed 2021-07-09 by SergeySlice

juanpc2018, 2021-09-09:
i found some problems/sollutions, for those interested:

Windows9.1 when installed in old machines 2010, has a BIOS Boot BCD partition,
Windows8.1 when installed in Newer machines, has UEFI Boot BCD,

using Windows Recovery USB, i can search, look, select, and delete the boot partition, 
format in Fat32,
and create a New bcdboot UEFI or BIOS or Both, /ALL
but /ALL does Not work well in a Real Mac, Macs are UEFI, but load Legacy BIOS partitions. 
and UEFI does Not work in older HW 2010 like x58 boards.

BUT.. i found something amazing... The Opposite:
a Bootloader called DUET
https://www.win-raid.com/t3286f50-Guide-NVMe-boot-for-systems-with-legacy-BIOS-and-UEFI-board-DUET-REFIND.html
a fork of Clover + rEFInd

Allows to Boot UEFI OS boot partition, in Old Bios Hardware, Amazing.!.
having all the advantages of UEFI,
over 4TB hdd, NVMe boot, GPT, etc...

Example:
Windows8.1x64 has NVMe drivers, R&W NVMe, But...
Old Bios cannot Boot NVMe, Nor GPT.
DUET Bootloader allows to break the 600MB/s SATA-III barrier in old Bios boards,
with M.2 to PCIe_x4 adapter like:
https://www.startech.com/en-us/hdd/pex4m2e1
having 1500MB/s R&W in Old Boards with PCIe v2.0
or 2500MB/s with special M.2 to PCIe adapter that use all x16 channels in PCIe v2.0
like Sonnet 4x4 M.2
HighPoint 8xM.2 has x16 PCIe but i dont know if joins all x16 channels when used as PCIe v2.0
Speed depends on available CPU lanes, having other cards eats lanes.
Old = year 2010, x58 intel chipset, Nehalem & Westemere CPUs socket 1366,
AMD Opteron 6*00 series CPUs G34 socket.
Server Boards AMD & Intel from Tyan and Supermicro
and similar PCs with PCIe v2.0 SATA 3Gbps 6Gbps, DDR3 memory..
as far is i know and tested, No Old board can boot UEFI partition, or GPT, or NVMe.
DUAL USB allows to boot UEFI in Old Non UEFI Boards.

Windows8.1x64
MBR + UEFI works Ok.
No need to convert to GPT if booting on a PC.
on MacPro5,1 2010 "has UEFI" and creates a conflic,
having dual Bootloadres with MBR.
to Boot UEFI... requires to change the bootloader,
when Win8.1x64 is installed on a Bios machine or Legacy mode in a new machine, installer creates a Bios bootloader, that needs to be erased, to create a New Windows bootloader with UEFI or UEFI+Bios.

Oversimplified Steps:
create a Windows Recovery USB drive,
boot from recovery USB.
diskpart
select boot partition, Not the c:\Windows drive.
delete boot partition,
Do Not type Clean,
Clean Erase the whole HDD/SSD..
Do Not type Clean.
format boot partition as fat32,
exit diskpart
bcdboot /f ALL to install Bios & UEFI bootloader, or UEFI or Bios, to create a only UEFI or only Bios boot partition.
done.

reboot from DUET USB
msinfo32.exe in cmd / terminal, allows to see if Bios Mode: Legacy or UEFI.
there are many Online video tutoriasl:
https://www.youtube.com/watch?v=u_ROQyKBDRo
when Windows boots Ok from UEFI, 1st time takes time coming from Bios Legacy boot.
then move Windows to a New NVMe drive...
creating a Restore Drive to an external 1TB USB3.0 drive..
turn off, remove the source HDD/SSD,
Boot Duet
Boot Windows Recovery USB
"2x different USB sticks",

Windows Recovery USB, can only restore a Bios Windows in a Bios machine,
and UEFI Windows in a UEFI machine.

Does Not allow to boot Bios, and restore UEFI Windows.
Does Not allow to boot UEFI, and restore Bios Windows.
Restore Windows from 1tb external Drive to New NVMe drive.

Turn Off,
remove Windows USB,
Reboot DUET,   Boot from NVMe.

Activate Windows again,

if its a Retail Version allows unlimited activations.
OEM Windows Only allows 1 Activation, to avoid people stealing windows in large companies.
there are other paid options using: EaseUS Partition Master.

P.D. DUET2020 USB creator does Not create USB in Win8.1,
everything works exept creating a modified MBR Boot USB,
requires W10x64 to create the modified MBR Boot USB.
but then Duet USB works OK with Win8.1x64,
only has problems creating the USB. then works ok.
W10 can be a VirtualMachine like: Orable VirtualBox, VMWare, Parallels, etc...
USB should work ok.

IF want to convert MBR to GPT:
needs to install the MBR2GPT tool + the other steps.
https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/mbr2gpt-tool-test-guidance
https://docs.microsoft.com/en-us/windows/deployment/mbr-to-gpt
https://docs.microsoft.com/en-us/windows-hardware/get-started/adk-install
https://developer.microsoft.com/en-us/windows/downloads/sdk-archive/

AT YOUR OWN Risk,
but if you create a Windows Restore drive, BEFORE
there is No risk.

insert NTFS 1TB drive,
Control Panel / System & Security / File Hystory.
Create USB
Create Backup
Done,