CloverHackyColor/CloverBootloader #424: GetOS  Wrong value if updating from an older OS to Monterey (tested from 11.4 and 11.5 beta3)

fabiosun, 2021-06-28:
as requested here : 
https://github.com/CloverHackyColor/CloverBootloader/issues/283#issuecomment-869458846
I open a new thread with a pretty same old problem in GetOS detection

SergeySlice, 2021-12-19:
I see no wrong GetOS detection. Check again and provide actual debug.log.

fabiosun, 2021-12-19:
Thank you SergeySlice
Problem is there from an old 5124 version and was present also in latest 5142
As soon I can I put back an high sierra 12.13.6 OS to update to a new one and see
From which commit you have solved this problem?

SergeySlice, 2021-12-25:
There are too many changed in sources related to switch from C to C++.
With latest Clover and latest macOS I see good detection.

fabiosun, 2022-02-06:
> I see no wrong GetOS detection. Check again and provide actual debug.log.
[image]
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/8009999/Archive.zip)

Installed on a clean disk High Sierra 
updated fine to Mojave and then to Catalina
trying to update to Big Sur from Catalina
First stage of updating in OSX Catalina is ok
the on Clover Boot menu I have two options
Boot from macOS OS data and boot from macOS
first one hangs 
second one boot in Catalina 
in debug log of first one I see OS X detected as 10.5.2.... but I started with this actual update:
[image]
a bit different problem but always a getOS detection problem here

closed 2022-02-10 by SergeySlice