CloverHackyColor/CloverBootloader #426: CustomLogo is not working

koroki, 2021-07-04:
Since I updated CLOVER into 5136, the CustomLogo option is not working in Linux or Windows.

In older versions, clover splashes the system's UEFI logo (aka BGRT) when I load Linux or Windows when I used CustumLogo None. But I could change for any logo. Now, when I load any of this systems, clover splashes a black screen.
In fact, for windows no logo is used when it is charged, only the typical spin, when before I always have the BGRT. 

Is there any issue or any option was changed?

SergeySlice, 2021-07-06:
In what Clover version this was ever worked?

koroki, 2021-07-08:
After the update, I was checking that 5084 leave the UEFI logo for windows.
But I cannot find the version that the UEFI logo was kept for Linux...

SergeySlice, 2021-07-09:
Starting Windows you will see Windows logo if UEFI boot and nothing if legacy boot.

koroki, 2021-07-11:
I'll recheck, thank you!

SergeySlice, 2021-07-26:
abandoned

closed 2021-07-26 by SergeySlice

koroki, 2022-09-20:
I know that it is late, but I can confirm that CustomLogo/Apple, works until version 5115 for linux kernel. In the newer versions, this option is not working and splash a black screen for 1/3 seconds. 

I copy my config:
`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Boot</key>
	<dict>
		<key>CustomLogo</key>
		<string>Apple</string>
		<key>Debug</key>
		<false/>
		<key>DefaultVolume</key>
		<string>LastBootedVolume</string>
		<key>Timeout</key>
		<integer>3</integer>
	</dict>
	<key>GUI</key>
	<dict>
		<key>Custom</key>
		<dict>
			<key>Entries</key>
			<array>
				<dict>
					<key>Arguments</key>
					<string>initrd=\amd-ucode.img initrd=\initramfs-linux.img root=UUID=xxxxxxxxxxxxxx rw quiet loglevel=3 rd.systemd.show_status=auto rd.udev.log_level=3 splash vt.global_cursor_default=0 </string>
					<key>Image</key>
					<string>os_arch</string>
					<key>Path</key>
					<string>vmlinuz-linux</string>
					<key>Title</key>
					<string>ArchLinux</string>
					<key>Type</key>
					<string>Linux</string>
					<key>Volume</key>
					<string>EFI</string>
					<key>VolumeType</key>
					<string>Internal</string>
				</dict>
				<dict>
					<key>CustomLogo</key>
					<string>Apple</string>
					<key>Image</key>
					<string>os_win</string>
					<key>Title</key>
					<string>Windows10</string>
					<key>Type</key>
					<string>Windows</string>
				</dict>
			</array>
		</dict>
		<key>Mouse</key>
		<dict>
			<key>Enabled</key>
			<false/>
		</dict>
		<key>Scan</key>
		<false/>
		<key>Theme</key>
		<string>mytheme</string>
	</dict>
</dict>
</plist>
`

koroki, 2022-09-20:
I can also confirm that the BootBgColor does not work from version 5111 (5110 is the last one that works). Sadly, 5110 does not have a complete support for theme.

koroki, 2022-09-20:
The "problem" happens with Linux (arch linux kernel) and with Windows10.

koroki, 2022-10-11:
@SergeySlice can you help me?