CloverHackyColor/CloverBootloader #428: Hd4000 patch software for clover under Mac OS 12

mjs520, 2021-07-05:
Can we develop a patch for clover hd4000 under Mac OS 12? OC's old patch doesn't work

mjs520, 2021-07-05:
[image]

mjs520, 2021-07-05:
[image]

z3DD3r, 2021-07-06:
Patch for HD4000 is not related to clover or any other boot loader. You can use OpenCore Legacy Patcher to install required kexts into system. Then u need to disable SIP (CsrActiveConfig=0xFEF in case of Clover) and add amfi_get_out_of_my_way=1 flag into boot options.

SergeySlice, 2021-07-06:
Nothing bootloader specific.

closed 2021-07-06 by SergeySlice