CloverHackyColor/CloverBootloader #434: Unable to build clover with ENABLE_SECURE_BOOT argument

Leopold702, 2021-07-20:
Hi,

I'm trying to build Clover with support for Secure Boot. If understood correctly from the posts on several forums, Secure Boot support isn't included into release version of the Clover and you can get one only by building it yourself. I'm trying to do this as follows:

1. Clone Clover repository:

`git clone --recursive https://github.com/CloverHackyColor/CloverBootloader.git`

2. CD to the cloned folder:

 `cd CloverBootloader/`

3. Build clover with ENABLE_SECURE_BOOT argument:

`./ebuild.sh -D ENABLE_SECURE_BOOT`

After executing the last command for some time building process seems to be running fine, however eventually it fails with following lines:

```
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/securevars.cpp:48:
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/err.h:124:
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/bio.h:69:
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/crypto.h:535:26: error: unknown type name 'FILE'
void CRYPTO_mem_leaks_fp(FILE *);
                         ^
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/securevars.cpp:48:
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/err.h:124:
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/bio.h:634:17: error: unknown type name 'FILE'
BIO *BIO_new_fp(FILE *stream, int close_flag);
                ^
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/bio.h:706:17: error: unknown type name 'FILE'
int BIO_dump_fp(FILE *fp, const char *s, int len);
                ^
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/bio.h:707:24: error: unknown type name 'FILE'
int BIO_dump_indent_fp(FILE *fp, const char *s, int len, int indent);
                       ^
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/securevars.cpp:48:
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/err.h:127:
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/lhash.h:186:33: error: unknown type name 'FILE'
void lh_stats(const _LHASH *lh, FILE *out);
                                ^
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/lhash.h:187:38: error: unknown type name 'FILE'
void lh_node_stats(const _LHASH *lh, FILE *out);
                                     ^
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/lhash.h:188:44: error: unknown type name 'FILE'
void lh_node_usage_stats(const _LHASH *lh, FILE *out);
                                           ^
In file included from /Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/securevars.cpp:48:
/Users/test/CloverBootloader/rEFIt_UEFI/entry_scan/../../Library/OpensslLib/openssl-1.0.1e/include/openssl/err.h:140:10: fatal error: 'errno.h' file not found
#include <errno.h>
         ^~~~~~~~~
8 errors generated.
make: *** [/Users/test/CloverBootloader/Build/Clover/RELEASE_XCODE8/X64/rEFIt_UEFI/refit/OUTPUT/entry_scan/securevars.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/Users/test/CloverBootloader/Build/Clover/RELEASE_XCODE8/X64/rEFIt_UEFI/refit]

build.py...
 : error F002: Failed to build module
	/Users/test/CloverBootloader/rEFIt_UEFI/refit.inf [X64, XCODE8, RELEASE]

- Failed -
Build end time: 08:56:51, Jul.20 2021
Build total time: 00:02:39
```

I'm not sure if this is a bug or I'm doing something wrong, but it worth noting that if I run just `./ebuild.sh`, without any arguments, then build completes successfully.

NontTheKiller, 2021-11-02:
Get the same error when trying to build clover with apple secure boot

WyndRyder, 2023-10-28:
Just thought I'd leave this here. I, too, am having issues trying to build signed clover (from Ubuntu). But in my attempt to solve my issues, I came across something.

Here are some steps to troubleshoot and potentially fix these issues:

    **Check Build Environment:** Ensure that the build environment is set up correctly. Given that the path seems to suggest macOS (RELEASE_XCODE8), verify that Xcode and its command-line tools are installed correctly.

    **Update OpenSSL:** The errors seem to originate from the openssl-1.0.1e version, which is a bit dated. Consider updating to a newer version of OpenSSL or checking if the Clover project has updates that address compatibility with newer versions of OpenSSL.

    **Include Standard Headers:** If for some reason standard headers are not being included, you might need to explicitly specify their paths. You can usually do this with compiler flags like -I followed by the path to the headers.

    **Clean and Rebuild:** Sometimes build artifacts from previous builds can cause issues. Consider cleaning the build directory and starting a fresh build.

    **Alternative Toolchains:** If you are not tied to Xcode, they could try building using GCC or another supported toolchain. Sometimes changing the compiler can sidestep compatibility issues.

SergeySlice, 2024-01-13:
Resolved in 5157.

closed 2024-01-13 by SergeySlice