CloverHackyColor/CloverBootloader #436: No visible entries for APFS on NVMe

d0plr, 2021-08-21:
Hi,
I cloned my working SATA SSD to a brand new Samsung 970 Evo Plus NVME using CCC, and copied the contents of the old EFI partition to the new one.
The new disk boots to Clover, but then do not show the icons for its APFS partitions (only those from other disks). Clover seems to detect the NVMe partitions just fine but decides to hide them.  Can you see what is causing this ?

[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/7025631/preboot.log)
[config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/7025632/config.plist.txt)

Regards

SergeySlice, 2021-09-03:
Your partitions are hidden
~~~~
6:874  0:001  - [18]: 'Preboot', hidden
6:874  0:000  - [19]: 'MacOS - Données', hidden
6:874  0:000  - [20]: 'MacOS', hidden
6:874  0:000  - [21]: 'Recovery', hidden

~~~~

See config-> Hide

d0plr, 2021-09-05:
Yes sorry, and thanks for looking into it. This is not a Clover issue, but here is what I think happened in case this help others with the same problem (web searches found some, but without resolutions).

The problem was the "\EFI\BOOT\BOOTX64.EFI" entry in the Hide section of the config. I do not remember why I had it (probably a copy & paste I did when I first installed Clover).  Somehow my old SATA disks do not have that file, while my new NMVe does. I guess that depends which macOS version was used to initialize them.  The solution is to remove that entry.

closed 2021-09-05 by d0plr