CloverHackyColor/CloverBootloader #439: Big Sur update from catalina errors as soon as it is started

eclair4151, 2021-08-29:
i am using the latest version of clover

I already have Catalina installed, but when i run the installer for big sur and boot from the data partition in verbose i am met with the following error. Does anyone know what might be causing this, or if trying to install via a bootable usb would work better?

Mobo: Asus Z390 pro
Proc: i9-9900k
Gpu: Radeon VII

Error: https://i.imgur.com/98sNDXL.jpeg

SergeySlice, 2021-09-03:
It may be CFGLock.
Provide debug.log if you want any help.
But this is not a clover issue.

akisame-reiu, 2021-09-04:
check CFG-Lock and Quirks section. Remember to tick KernelXCPM if your bios doesn't have CFG-Lock option. Also check the quirks by following config for CoffeeLake desktop on Dortania website