CloverHackyColor/CloverBootloader #444: How did I boot without Clover installed ?

mjur3191, 2021-09-17:
Config
DELL T7810 (TPM2.0), Samsung 980 NVME, Windows 10 21H

Installed Clover EFI using the Utility (5139 maybe) on SanDisk UltraFit. Fresh parts and fresh install. Removed USB stick. 

UEFI set to Clover originally, then Windows Boot Manager second

Boots from NVME without assistance .... how is this possible ?

SergeySlice, 2021-09-30:
It depends on NVMe supported by the BIOS. Else boot from NVMe is not possible. Clover will started only after the drive started.

mjur3191, 2021-09-30:
Dell T7810 has no known direct support for NVME boot. 

Bios settings are 
Mode: UEFI 
Priority: Windows boot manager

that’s all all. There is no other supporting volume. From where came the code to directly boot the NVME ? 

During installation of windows it is possible for the Windows ISO USB drive to see the NVME drive. On completion it will not boot.

if anyone will Google this topic it is well reported in forums inside Dell. Many customers are confused till it is made obvious they need newer motherboard. 

Did a patch file come from clover into boot manager or did Microsoft change version 21H so that it can achieve the same result as clover or does it have something with TPM2.0 ? I don’t know.

mjur3191, 2021-10-03:
Ok I have tested the variable. With the Dell Precision workstation T7810 and T5810 it is possible to boot NVME without Clover EFI if the TPM 2.0 is ON and ENABLED 

the bios must be in UEFI boot, I used RAID mode, first boot device is Windows Boot Manager 

With this configuration it is possible to remove Clover boot device. 

I have no explanation for this outcome. It is proven on 2 machines. One thing I must emphasise the TPM is owned by the OS on the NVME. If this matters I don’t know. 

Another thing, after bios update reset all bios to default and configure as above. Something about bios default reset improves the function.