CloverHackyColor/CloverBootloader #445: scan entries blocking on intel S1200V3RP

idreamerhx, 2021-09-18:
hi,

uefi boot in usb disk. tried r5139 r5121 r5118 ... all shows "scan entries" without any following actions.
i tried on a b360m board the usb disk with r5139 works.

it is realy hard to config clover to save debug.log. i only successfully configed debug in r5139, the log shows the last line "applesmc"

idreamerhx, 2021-09-23:
3:633  0:006  PlatformFeature will not set in SMBIOS
3:640  0:006   Dell variables
3:650  0:009  === [ Found DSDT tables ] =======================
3:665  0:015  === [ ScanVolumes ] =============================
3:671  0:006  Found 9 volumes with blockIO
3:678  0:006  - [00]: Volume: PciRoot(0x0)\Pci(0x1D,0x0)\USB(0x1,0x0)\USB(0x4,0x0)
3:692  0:014          Result of bootcode detection: bootable unknown (legacy)
3:704  0:011  - [01]: Volume: PciRoot(0x0)\Pci(0x1D,0x0)\USB(0x1,0x0)\USB(0x4,0x0)\HD(1,MBR,0x1A0465B4,0x3F,0x64000)
3:718  0:014          Result of bootcode detection: bootable unknown (legacy)
3:726  0:007          This is SelfVolume !!
3:732  0:006  - [02]: Volume: PciRoot(0x0)\Pci(0x1D,0x0)\USB(0x1,0x0)\USB(0x4,0x0)\HD(2,MBR,0x1A0465B4,0x6403F,0x713FC1)
3:747  0:014  - [03]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)
3:761  0:013  - [04]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)\HD(1,GPT,5E704E60-8A09-44F9-BEFF-C226D12BB795,0x800,0x96000)
3:775  0:014          Result of bootcode detection: bootable Windows (win)
3:785  0:009  - [05]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)\HD(2,GPT,ED7128B0-BC77-4FBB-AEAC-72EDCEA3A90B,0x96800,0x40000)
3:859  0:074          Result of bootcode detection: bootable unknown (legacy)
3:866  0:006  - [06]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)\HD(3,GPT,B8FA10FC-C1D8-4B90-BCFA-EC59D9463F99,0x100028,0x7800000)
3:881  0:014          Result of bootcode detection: bootable Windows (vista,win)
3:887  0:006  - [07]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)\HD(4,GPT,6761BD92-5CFC-4241-B373-DFAC348F2FAC,0x7900028,0x2000000)
3:901  0:013  - [08]: Volume: PciRoot(0x0)\Pci(0x1F,0x2)\Sata(0x0,0xFFFF,0x0)\HD(5,GPT,2EFA118D-C63F-4144-B572-A90F4F369906,0x9900800,0xB200800)
3:915  0:013          Result of bootcode detection: bootable Windows (vista,win)
3:921  0:006  No AudioIoProtocols, status=Not Found
3:928  0:007  === [ InitXTheme ] ==============================
3:935  0:006  use night theme
3:942  0:006   using embedded theme
3:963  0:021  sound play end with status=Device Error
3:971  0:008  Using embedded font
3:978  0:006  theme inited
3:984  0:006  Chosen embedded theme
3:994  0:009  after NVRAM boot-args=
4:001  0:006  === [ Dump SMC keys from NVRAM ] ================
4:007  0:006  no AppleSMC protocol

same as:
#224

SergeySlice, 2021-09-30:
I have AppleSMC protocol and see
~~~~
4:612  0:000  === [ Dump SMC keys from NVRAM ] ================
4:612  0:000  found AppleSMC protocol
4:632  0:019     fakesmc-key-#KEY-ui32:00 00 00 10 
4:633  0:001     fakesmc-key-$Adr-ui32:00 00 03 00 
4:634  0:001     fakesmc-key-$Num-ui8:01 
4:635  0:001     fakesmc-key-RMde-char:41 
4:639  0:003     fakesmc-key-RBr -ch8*:6A 39 35 6A 39 35 61 6D 
4:640  0:001     fakesmc-key-EPCI-ui32:08 C0 F0 00 
4:641  0:001     fakesmc-key-REV -ch8*:02 34 0F 00 00 03 
4:642  0:001     fakesmc-key-RPlt-ch8*:6A 39 35 00 00 00 00 00 
4:643  0:001     fakesmc-key-BEMB-flag:00 
4:644  0:001     fakesmc-key-BATP-flag:00 
4:646  0:001     fakesmc-key-BNum-ui8:00 
4:647  0:001     fakesmc-key-MSTc-ui8:00 
4:648  0:001     fakesmc-key-MSAc-ui16:00 00 
4:649  0:001     fakesmc-key-MSWr-ui8:00 
4:650  0:001     fakesmc-key-MSFW-ui8:01 00 
4:652  0:001     fakesmc-key-MSPS-ui16:00 03 
4:675  0:023  Registered 16 SMC keys
4:677  0:001  === [ ScanLoader ] ==============================

~~~~
I may propose you to install SMCHelper.efi if you are using FakeSMC.kext or VirtualSMC.efi if you are using VirtualSMC.kext.
Also the problem may be in ScanLoader for you if you installed NTFS.efi and have very large Windows partition. Wait half an our for the scan to complete or just erase NTFS.efi

There may be other reasons for the issue.