CloverHackyColor/CloverBootloader #454: Can't build Clover since first r_5140 release (commit 83b9323 on Sep 29, 2021) nor with r_5140.1

matxpa, 2021-10-01:
Hi Slice & Jief.

I can't build Clover (X64, XCODE8, RELEASE) with Xcode Version 13.0 (13A233) (clang-1300.0.29.3) nor with Xcode 12.5.
The last release i could make was Clover r5139 (583e2dd) on 20210924 (X64, XCODE8, RELEASE).

Error lines (end of log)
...
[CPP] AcpiPatcher
[SLINK] EfiLoader
[SLINK] BdsDxe
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/AcpiPatcher.cpp:26:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/AcpiPatcher.h:11:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../Platform/Volume.h:14:
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../libeg/libeg.h:139:15: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
typedef struct {
              ^
               REFIT_DIR_ITER
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../libeg/libeg.h:140:36: note: type is not C-compatible due to this default member initializer
  EFI_STATUS          LastStatus = 0;
                                   ^
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../libeg/libeg.h:144:3: note: type is given name 'REFIT_DIR_ITER' for linkage purposes by this typedef declaration
} REFIT_DIR_ITER;
  ^
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/AcpiPatcher.cpp:29:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/smbios.h:15:
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../Platform/cpu.h:344:15: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
typedef struct {
              ^
               CPU_STRUCTURE
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../Platform/cpu.h:346:49: note: type is not C-compatible due to this default member initializer
  UINT32                  CPUID[CPUID_MAX][4] = {{0}};
                                                ^~~~~
/Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../Platform/cpu.h:394:3: note: type is given name 'CPU_STRUCTURE' for linkage purposes by this typedef declaration
} CPU_STRUCTURE;
  ^
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/AcpiPatcher.cpp:32:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Platform/../Settings/SelfOem.h:12:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Settings/../Settings/ConfigManager.h:11:
In file included from /Users/meunier/CloverBootloader/rEFIt_UEFI/Settings/ConfigPlist/ConfigPlistClass.h:19:
/Users/meunier/CloverBootloader/rEFIt_UEFI/Settings/ConfigPlist/../../include/Pci.h:38:14: error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage]
typedef union {
             ^
              pci_dev_t
/Users/meunier/CloverBootloader/rEFIt_UEFI/Settings/ConfigPlist/../../include/Pci.h:39:22: note: type is not C-compatible due to this default member initializer
  uint32_t    addr = 0;
                     ^
/Users/meunier/CloverBootloader/rEFIt_UEFI/Settings/ConfigPlist/../../include/Pci.h:41:3: note: type is given name 'pci_dev_t' for linkage purposes by this typedef declaration
} pci_dev_t;
  ^
3 errors generated.
make: *** [/Users/meunier/CloverBootloader/Build/Clover/RELEASE_XCODE8/X64/rEFIt_UEFI/refit/OUTPUT/Platform/AcpiPatcher.obj] Error 1

build.py...
 : error 7000: Failed to execute command
	make tbuild [/Users/meunier/CloverBootloader/Build/Clover/RELEASE_XCODE8/X64/rEFIt_UEFI/refit]

build.py...
 : error F002: Failed to build module
	/Users/meunier/CloverBootloader/rEFIt_UEFI/refit.inf [X64, XCODE8, RELEASE]

- Failed -
Build end time: 12:14:28, Sep.30 2021
Build total time: 00:00:54

Full Log [Clover_r5140_Compil_XcodeV13.0_KO.txt](https://github.com/CloverHackyColor/CloverBootloader/files/7265301/Clover_r5140_Compil_XcodeV13.0_KO.txt)

Regards

matxpa, 2021-10-01:
The 3 next components were modified in Commit 2c829b6 on Sep 28, 2021 "Create XBool to avoid nasty side effect when assigning int to bool."
CloverBootloader/rEFIt_UEFI/libeg/libeg.h (lines 139, 140 and 144)
CloverBootloader/rEFIt_UEFI/Platform/cpu.h (lines 344, 346 and 394)
CloverBootloader/rEFIt_UEFI/include/Pci.h (lines 38, 39 and 41)

stevezhengshiqi, 2021-10-01:
See https://github.com/CloverHackyColor/CloverBootloader/issues/453

matxpa, 2021-10-02:
Hi Steve.

I'm sorry but the 3 errors aren't "misleading-indentation" but "Wnon-c-typedef-for-linkage"

error: anonymous non-C-compatible type given name for linkage purposes by typedef declaration; add a tag name here [-Werror,-Wnon-c-typedef-for-linkage] for libeg.h, cpu.h and Pci.h.

I'll try with Slice's answer (switch to avoid indentations) and report.

Thanks for your answer
Regards

matxpa, 2021-10-02:
I added switch "-Wno-misleading-indentation" in content of the var XCODE8_COMMON_FLAGS in file ~/CloverBootloader/BaseTools/Conf/tools_def.template and the result of "./buildme XCODE8" (option 1) is (the same) : 3 errors for "non-c-typedef-for-linkage" in libeg.h, cpu.h and Pci.h.

But may be I'm wrong on the choice of var XCODE8_COMMON_FLAGS.

Regards

matxpa, 2021-10-02:
-Wno-non-c-typedef-for-linkage added to XCODE8_COMMON_FLAGS (-Wno-misleading-indentation removed) in clang config file ~/CloverBootloader/BaseTools/Conf/tools_def.template and "./buildme XCODE8" (option 1)
...
- Done -
Build end time: 21:05:54, Oct.02 2021
Build total time: 00:06:36

  -> CloverX64-RELEASE_XCODE8-20210929163255-83b9323-dirty.efi
  adding: ~/CloverBootloader/Build/CloverX64-RELEASE_XCODE8-20210929163255-83b9323-dirty.efi (deflated 53%)

But is this CloverX64.efi XCODE8 build reliable ?

SergeySlice, 2021-10-03:
I have older clang
~~~~
Apple clang version 11.0.0 (clang-1100.0.33.12)
Target: x86_64-apple-darwin21.1.0

~~~~
so I didn't check new C++20 warnings.
I have to propose use -Wno-non-c-typedef-for-linkage

Try to change the sources such a way
~~~~
typedef struct  **REFIT_DIR_ITER**{
  EFI_STATUS          LastStatus = 0;
  const EFI_FILE     *DirHandle = NULL;
  XBool               CloseDirHandle = false;
  EFI_FILE_INFO      *LastFileInfo = NULL;
} REFIT_DIR_ITER;

~~~~
Without the -Wno-....

SergeySlice, 2021-10-03:
Same for CPU_STRUCTURE and for pci_dev_t.

matxpa, 2021-10-03:
Hi Slice.

Thanks.

I applied the solution you proposed (ie. add a tag on each Struct or Union typdef) and Clover's compilation was successful with buildme XCODE8 on both options : 1) build Clover and 8) build all.

What I did :
1) remove switch "-Wno-non-c-typedef-for-linkage"  from XCODE8_COMMON_FLAGS variable in clang config file ~/CloverBootloader/BaseTools/Conf/tools_def.template
2) modify the others components (4 Header files, I had to modify rEFIt_UEFI/Platform/HdaCodecDump.h too)

CloverBootloader/rEFIt_UEFI/include/Pci.h (line 38)
		typedef union pci_dev_t {
		...
		} pci_dev_t;

CloverBootloader/rEFIt_UEFI/libeg/libeg.h (line 139)
		typedef struct  REFIT_DIR_ITER {
		...
		} REFIT_DIR_ITER;

CloverBootloader/rEFIt_UEFI/Platform/cpu.h (line 344)
		typedef struct CPU_STRUCTURE {
		...
		} CPU_STRUCTURE;

CloverBootloader/rEFIt_UEFI/Platform/HdaCodecDump.h (line 35)
		typedef struct HdaWidget {
 		...
		} HdaWidget;               

So thank you very much for your suitable solution
matxpa

matxpa, 2021-10-05:
Hi Slice.

Your two latest commits 198a8e3 and 26f6050 resolve the issue.

Clover r5140 (X64, XCODE8, RELEASE) commit 26f6050 successfully complied with Xcode 13.0 (13A233) (clang-1300.0.29.3) 

MBP MacBookPro11,5 Big Sur 11.6.1 ⌘  Powered by Clover revision: 5140 (master, commit 26f605062)

So this issue can be closed.

Thanks again

closed 2021-10-05 by matxpa