CloverHackyColor/CloverBootloader #472: Questions about adding rdrand patch to clover

mjs520, 2021-11-03:
<dict>
				<key>Comment</key>
				<string>Monterey RDRAND instruction #1</string>
				<key>MatchOS</key>
				<string>12</string>
				<key>Disabled</key>
				<false/>
				<key>Procedure</key>
				<string>_work_interval_port_type_render_server</string>
				<key>RangeFind</key>
				<integer>3900</integer>
				<key>Find</key>
				<data>D8fxc/sh8TnRc/WJyUiLlM3Q/f//</data>
				<key>Replace</key>
				<data>McmQkJAh8TnRc/WJyUiLlM3Q/f//</data>
			</dict>
			<dict>
				<key>Comment</key>
				<string>Monterey RDRAND instruction #2</string>
				<key>MatchOS</key>
				<string>12</string>
				<key>Disabled</key>
				<false/>
				<key>Procedure</key>
				<string>_panic_with_thread_context</string>
				<key>RangeFind</key>
				<integer>10100</integer>
				<key>Find</key>
				<data>D8fyc/uD4g+D+gd38w+3NEE=</data>
				<key>Replace</key>
				<data>MdKQkJCD4g+D+gd38w+3NEE=</data>
			</dict>

The patch kernel under clover kerneltopatch doesn't seem to work. Does clover lack any dynamic patch support? Thank you

mjs520, 2021-11-04:
ok

mjs520, 2021-11-04:
[image]
My CPU does not support rdrand

SergeySlice, 2021-11-04:
Users confirmed the patch working.

mjs520, 2021-11-04:
> Users confirmed the patch working.

At present, the feedback seems to be from OC users. Clover didn't find it successful. Thank you

mjs520, 2021-11-10:
[image]
The rdrand restriction was removed from 12.1 beta 2

closed 2021-11-10 by mjs520