CloverHackyColor/CloverBootloader #474: Clover r5142 broke booting

nsby494, 2021-11-15:
When I clicked on the Install macOS Catalina installation disk, the notebook was in a black screen state with backlight and no characters displayed. I turned on the debug mode and turned on -V, the notebook is Fujitsu T732 i5 3320
The following is my installation log, can you help me check where the problem is, thank you!
[2021-11-15_02-21_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/7535479/2021-11-15_02-21_BOOTX64.EFI.log)
[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/7535480/config.plist.zip)

SergeySlice, 2021-11-28:
I may recommend to check something
1. Exclude AudioDxe.efi. Sometimes it causes problems.
2. SMCBatteryManager.kext is not compatible with FakeSMC. Use VoodooBatterySMC.kext.
3. Set KernelPM=true, your settings is =false which is wrong as your BIOS locked.

SergeySlice, 2021-12-19:
abandoned

closed 2021-12-19 by SergeySlice