CloverHackyColor/CloverBootloader #475: How to get in touch regarding a security concern

zidingz, 2021-11-25:
Hey there!

I belong to an open source security research community, and a member (@dwisiswant0) has found an issue, but doesn’t know the best way to disclose it.

If not a hassle, might you kindly add a `SECURITY.md` file with an email, or another contact method? GitHub [recommends](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository) this best practice to ensure security issues are responsibly disclosed, and it would serve as a simple instruction for security researchers in the future.

Thank you for your consideration, and I look forward to hearing from you!

(cc @huntr-helper)

SergeySlice, 2021-12-25:
What about you? Is it somehow related to the Clover?
What kind of security may be if the project is open source?

JamieSlome, 2021-12-26:
@SergeySlice - you can view the report here:

https://huntr.dev/bounties/1f9a139d-61ff-476f-82f9-d62d85d99350/

SergeySlice, 2022-01-13:
> @SergeySlice - you can view the report here:
> 
> https://huntr.dev/bounties/1f9a139d-61ff-476f-82f9-d62d85d99350/

And what?
[image]

JamieSlome, 2022-01-14:
@SergeySlice - do you have an e-mail that we can send the report to, it won't require any login but gives you full access to the report.