CloverHackyColor/CloverBootloader #477: Clover 5142 Monterey 12.1 beta 3 (Black screen in login page)

fabiosun, 2021-11-26:
Monterey 12.01 is perfectly booting
Monterey 12.1 beta 3 no, black screen (system is there ma no signal on display)
Same attached EFI
[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/7606965/EFI.zip)

Logs:
[2021-11-26_06-57_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/7606966/2021-11-26_06-57_BOOTX64.EFI.log)
[2021-11-26_06-58_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/7606967/2021-11-26_06-58_BOOTX64.EFI.log)

SergeySlice, 2021-11-28:
Monterey 12.1 beta 3 booted fine
[image]
.
Your system is Ryzen, so ask Shunny for a set of Ryzen patches for the new system. It is not Clover issue.

closed 2021-11-28 by SergeySlice

fabiosun, 2021-11-28:
@SergeySlice 
It is not a Ryzen problem
Same patches work in Opencore
And your boot loader I think "fails" in 12.1 in New ResizeBar quirk!
I am one of the maintainer of AMD clover patches..so I have to ask to you :)

fabiosun, 2021-11-28:
I would like to explain better
And kernel patches are the same for Monterey 12.01 and latest 12.1 betas...

Shaneee, 2021-11-28:
> Monterey 12.1 beta 3 booted fine
> [image]
> .
> Your system is Ryzen, so ask Shunny for a set of Ryzen patches for the new system. It is not Clover issue.

Patches are fine. They are applied from Clover without issue but still black screen. Same patch set works without issues in OpenCore.

fabiosun, 2021-11-28:
[image]
problem solved!