CloverHackyColor/CloverBootloader #478: Move the wiki on sourceforge to github

ShortArrow, 2021-12-02:
move the wiki on sourceforge to github, I want to help you.

# Before

https://sourceforge.net/p/cloverefiboot/wiki/Home/

# After

https://github.com/CloverHackyColor/CloverBootloader/wiki

5T33Z0, 2021-12-19:
@ShortArrow That's why I created this: https://github.com/5T33Z0/Clover-Crate

ShortArrow, 2021-12-24:
@5T33Z0 Are you planning to merge [it](https://github.com/5T33Z0/Clover-Crate) into this repository?

SergeySlice, 2021-12-25:
What do you need to do this? Additional rights?

ShortArrow, 2021-12-25:
> What do you need to do this? Additional rights?

Which one are you talking about?  

I would be grateful if you could invite me to the cloverhackycolor organization.  But if it's essential, it's not.

I'm trying to confirm the direction before forking and starting work.  I think it's a waste of time to do something you don't want.

In any case, I think it is possible to fork and pull request.  Do @5T33Z0 do it in @5T33Z0 's repository, or I do it, or do nothing.

ShortArrow, 2021-12-25:
There are two things I want to know.

Is there a mechanism to automatically generate a document from source code? Or will it be adopted in the future? 

Do you think it's a good idea to add CloverCrate to the readme?

5T33Z0, 2021-12-25:
The clover crate repo has been mentioned in the changelog of a previous release. I think there's no real readme included clover in the clover package. The included docs in the efi folder are also really, really old. 

Since the way documentation has been handled so far is really poor, I think I will keep control of this repo. But everybody is welcome to fork it or contribute to it.

ShortArrow, 2021-12-25:
First of all, I think it's best to put the Crate URL as a semi-official in the clover README.  If possible, you may consider integrating repositories.

SergeySlice, 2021-12-26:
You are welcomed.

closed 2021-12-26 by SergeySlice