CloverHackyColor/CloverBootloader #480: How do I install?

LexiTheTT, 2021-12-02:
Howdie there, I'm very very new to the world of duel booting 'n the like. How do I install this?

LexiTheTT, 2021-12-02:
OH right, extra deets. I'm running ubuntu as my daily!

glnzglnz, 2021-12-04:
Have same question.  How do install this onto a USB stick?

florianb, 2021-12-12:
I found this introduction: https://wiki.archlinux.org/title/Clover

glnzglnz, 2021-12-12:
florianb - thanks, but I'm too much of a noob and don't know what to do with that info.
I don't have Linux.  Also, I was hoping to install clover onto a USB stick, boot my Win 10 Pro 64-bit PC from it, and use it to continue to boot the PC from a NVMe M.2 SSD inserted into an adapter in my PCIe 4x slot, when my PC is too old to boot from PCIe.  (The NVMe will have a clone of my current OS, apps, etc.)
Any ideas how to get started on that?  How to install Clover onto the USB stick?
Thanks.

florianb, 2021-12-12:
Thats a good question you might try downloading the ISO from the releases page and then might try to put that on a USB stick using f.e. https://unetbootin.github.io/

But no guarantees - i am trying to revamp a legacy iMac.

glnzglnz, 2021-12-12:
florianb - thanks again.  I looked at the releases page and did not find an iso.  
There's something in one of the downloads called CloverISO-5142.tar, but I'm not sure what to do with that.
Also, my final USB stick will be booting a Windows machine.  Maybe the stick itself can be in Linux, but I'm not sure about that.

SergeySlice, 2021-12-19:
And what is the issue? You can't find any instruction somewhere in internet? They are billion.

closed 2021-12-19 by SergeySlice