CloverHackyColor/CloverBootloader #484: ProvideCurrentCPUInfo quirk [Implementation request]

fabiosun, 2021-12-18:
Hi dev
From Opencore 070 as you can see in this pictures:

[image]

a new quirk was introduced in OpenCore Functionality
This quirk in this day assumes an important interest because it adds a preliminary support for new Alder lake CPU
personally I am not interested in this new cpu family..
But it could be very useful to implement it in Clover because it helps AMD CPU users to avoid some Kernel patches
In my opinion this quirk, if well implemented, could be also useful to avoid some problems AMD users have with latest Monterey 12.1 (dock speed animation and also weird and lower GPU benchmark)
Obviously with Opencore release or beta AMD users don't have this problem

thank you for your attention on this problem

SergeySlice, 2021-12-20:
This patch actually is Kernel patch and can be done by config.plist, Just to know what and where.
For AMD users the quirk is undesirable because of interfere with other patches for AMD CPU.

fabiosun, 2021-12-20:
> This patch actually is Kernel patch and can be done by config.plist, Just to know what and where.
> For AMD users the quirk is undesirable because of interfere with other patches for AMD CPU.

No @SergeySlice 
With that quirk Amd users can skip to use 3 kernel patches we have to use to boot in OSX with CLOVER bootloader.
By the way, We are also using an old Cpu topology kernel patch because new one included in that quirk is not compatible with clover bootloader masking feature
Take a look to Amd-kernel patches GitHub if you like to give an hand to us and to give a chance to AMD users to use also your bootloader
Thank you

SergeySlice, 2022-02-05:
The patch implemented in 5144. Test please and report.

fabiosun, 2022-02-05:
I thought I have closed also this issue
Sorry
It works perfectly

closed 2022-02-05 by fabiosun