CloverHackyColor/CloverBootloader #487: machdep.tsc.frequency on AMD CPU System - (maybe related to the miss of ProvideCurrentCPUInfo quirk)

fabiosun, 2021-12-27:
This problem is present also in old Clover Release (tested till R5139)

In OSX greater than 12.01 it produces many disasters as instability, double speed animation in dock o in YouTube , bad GPU performances, WIFi disconnection and so on)

Attached here a pic:
[image]

left is Clover terminal output and right is about Opencore boot loader output

Attached also 2 logs, first one is about a "working" OSX like Monterey 12.01:
[2021-12-26_09-19_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/7778966/2021-12-26_09-19_BOOTX64.EFI.log)

I say "working" because in Monterey 12.01 system works pretty fine and the only problem I see is a GPU performance loss of about 12% if compared with OpenCore boot loader 

Second one log is about 12.2 (it is similar or the same in 12.1 OS):
[2021-12-26_09-23_BOOTX64.EFI.log](https://github.com/CloverHackyColor/CloverBootloader/files/7778980/2021-12-26_09-23_BOOTX64.EFI.log)

12.1 is a disaster, with all problem said above

I attach here also the patches (old) we have to use in Clover Bootloader:
[patches.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/7779010/patches.plist.zip)

These above patches are included in a newer release in OpenCore Bootloader Kernel quirk
We can't use this new three patches because Clover manages in a different way masking parameters, and C3/ret does not work as it does in OpenCore

If you need others things to debug in a more proper way ask and I will try to answer as soon I can

Thank you

SavageAUS, 2022-01-19:
+1 for this issue as well. Really needs to be looked at deeper.

SergeySlice, 2022-01-28:
ProvideCurrentCpuInfo has no relation to AMD CPU.
You just need a correct set of kernel patches for AMD.

fabiosun, 2022-01-28:
No @SergeySlice sorry to contradict you
Patches are fine and we need of three additional patches in clover because we do not have that quirk:
**algrey - cpu_topology_sort -disable _x86_validate_topology** _Not mandatory for all AMD CPU)_
**tsc_init - remove Penryn check to execute default case** _mandatory for all AMD CPU_
**tsc_init - grab DID and VID from MSR** _mandatory for all AMD CPU_

```
`17. ProvideCurrentCpuInfo
Type: plist boolean
Failsafe: false
Requirement: 10.8 (10.14)
Description: Provides current CPU info to the kernel.
This quirk works differently depending on the CPU:
• For Microsoft Hyper-V it provides the correct TSC and FSB values to the kernel, as well as disables CPU topology validation (10.8+).
```
And with Clover Bootloader we have wrong TSC data from many time...in Monterey 12.2 it produces bad animation and many others problem
In previous os problem is there but produce only a worst GPU performances for many of us (about 10%)

Thank you for your time

fabiosun, 2022-01-29:
Grazie @SergeySlice your latest R5144 solve almost all problem AMD users add with Clover and TSC:
```
Last login: Sat Jan 29 12:26:21 on console
fabio@fabios-Mac-Pro ~ % sysctl -a | grep freq
hw.busfrequency: 100000000
hw.busfrequency_max: 100000000
hw.busfrequency_min: 100000000
hw.cpufrequency: 3700000000
hw.cpufrequency_max: 3700000000
hw.cpufrequency_min: 3700000000
hw.tbfrequency: 1000000000
machdep.tsc.frequency: 3700000000
fabio@fabios-Mac-Pro ~ % 
```

closed 2022-01-29 by fabiosun