CloverHackyColor/CloverBootloader #488: WARNING: NAME is not defined

ellaosx2007, 2021-12-29:
Clover v5143, boot shows a warning about "Name" is not defined in kextstopatch.
The warning points to new RDRAND entry.

SergeySlice, 2021-12-29:
There are no "Name" or "NAME"
~~~~
		<key>KernelToPatch</key>
		<array>
			<dict>
				<key>Comment</key>
				<string>Monterey RDRAND instruction #1</string>
				<key>MatchOS</key>
				<string>12</string>
				<key>Disabled</key>
				<false/>
				<key>Procedure</key>
				<string>_work_interval_port_type_render_server</string>
				<key>RangeFind</key>
				<integer>3900</integer>
				<key>Find</key>
				<data>D8fxc/sh8TnRc/WJyUiLlM3Q/f//</data>
				<key>Replace</key>
				<data>McmQkJAh8TnRc/WJyUiLlM3Q/f//</data>
			</dict>
			<dict>
				<key>Comment</key>
				<string>Monterey RDRAND instruction #2</string>
				<key>MatchOS</key>
				<string>12</string>
				<key>Disabled</key>
				<false/>
				<key>Procedure</key>
				<string>_panic_with_thread_context</string>
				<key>RangeFind</key>
				<integer>10100</integer>
				<key>Find</key>
				<data>D8fyc/uD4g+D+gd38w+3NEE=</data>
				<key>Replace</key>
				<data>MdKQkJCD4g+D+gd38w+3NEE=</data>
			</dict>
		</array>

~~~~

closed 2021-12-29 by SergeySlice

ellaosx2007, 2021-12-31:
[image]