CloverHackyColor/CloverBootloader #492: Upgraded to 5143 and getting BiosVersions issues during boot..

gateway, 2022-01-17:
Not sure whats changed but when I recently upgraded to the latest Clover and when I boot or run the clover config.plist validator it spits out a few warnings .. is this ok or do I need to change something with clover configurator?

```gateway@iMac ~ % ccpv /Volumes/EFI/EFI/CLOVER/config.plist
Warning: BiosVersion 'IM191.88Z.F000.B00.2004232100' is before than default ('IM191.88Z.F000.B00.2106222356') -> ignored. Dict '/SMBIOS:316'.
Warning: BiosReleaseDate '04/23/2020' is older than default ('06/22/2021') -> ignored. Dict '/SMBIOS:316'.
Warning: EfiVersion '1037.120.79.0.0' is older than default ('1554.140.20.0.0') -> ignored. Dict '/SMBIOS:316'.```

leomenassun, 2022-01-18:
Just open your config.plist with Clover Configurator, go to SMBIOS tab a mark the checkbox “firmware update only”. After this, open the dropdown smbioses’ list and chose the same smbios you already have. It will update information of your smbios, except you serial number. ☺️

SergeySlice, 2022-01-20:
It is OK

closed 2022-01-20 by SergeySlice