CloverHackyColor/CloverBootloader #494: CloverBootloader as NVME Controller / NVME SSD Disk firmware.

cederom, 2022-01-20:
Hello world :-)

* I have made NVME Controller ICYBOX IB-PCI224M2-ARGB PCI-E 4.0 x4 and NVME SSD Disk running on old Motherboard (ASUS M5A97 R2.0 with PCI-E 2.0 and latest available BIOS 2603 from 2015).
* I have made this configuration working with my FreeBSD AMD64 13-STABLE on ZFS `/` running from NVM SSD Disk. Booting with UEFI, CSM disabled. Bootloader and Kernel is loaded from onboard SATA Disk that then mounts `/` from the NVM.
* Linux is known to work in this setup. Windoze 10 installer booted from DVD can see the controller and disk.
* Problem is that BIOS / Motherboard does not see the NVM Controller nor Disk and so it cannot set it as boot device.

Question: is it possible to flash Clover EFI Bootloader into the NVM Controller or NVM SSD Disk as some sort of firmware that would allow booting directly from the NVM on a moterboard that does not see the controller and cannot on its own set the NVM as the boot device? :-)

succeedmr123, 2022-01-27:
I believe what you wanted to express is:

Can additional hardware interfaces be seen by Clover in order for storage devices to be booted from, and which method to use to connect it with Clover or the underlying BIOS?

e.g. you need NVME to be found over PCI-E
someone else want discrete USB 3.0 to be "seen" so that the devices connected to it be bootable from, same as NVME over PCI-E.

I got the same wish, on an older machine where even USB3.0 (Renesas/Nec) can not be used to boot storage devices natively from BIOS due to the lazy /inprecise implementation of the Laptop/BIOS Manufacturer

More precisely, the solution exists, and I believe  it is over additional "drivers" in UEFI form

Unless you feel like a BIOS hacker, and find a way to hack it into integrating hardware support into it and flashing it back into the machine (some ppl like hard challenges). In that case, better have BIOS chip direct flash tools, and a screwdriver to disassemble the machine to access it.
Hope it helps

cederom, 2022-01-27:
I have upgraded motherboard from ASUS M5A97 R2.0 to ASUS Sabertooh 990FX R2.0 and the new motherboard already has the NVM support in BIOS so I can boot from drive directly now :-)

Thanks @succeedmr123, this seems more tricky. I know the distinction between controller (PCI-E card) and device (disk). There are two possible scenarios:

1. Clover EFI Boot itself is a controller/device firmware. Here controller is a PCI-E card, while device is a disk drive. While UEFI firmware on the controller seems to be a standard, I am not sure if UEFI firmware on a disk is possible (especially when BIOS does not see the controller in the first place). That would imply the controller would have to load the firmware from a device and provide it as its own firmware to the BIOS.
2. Some sort of glue firmware that would UEFI boot the unknown device for BIOS and launch the Clover from EFI a partition. In that case we have small generic "glue" firmware loaded into the Controller that would allow it to boot of the disk, even if the controller is not visible to the BIOS. In this scenario we may enforce standard UEFI firmware boot, even if the card is not supported by BIOS, but as standard PCI-E it may expose the firmware, so the firmware then would detect a drive and chainload the boot from EFI partition.

I am not sure if PCI-E controller card (may provide UEFI firmware) is the same as external USB device (uses onboard USB controller that already is the PCI-E card itself). If both are possible then we have perfect situation (we talk about device firmware in that case). But PCI controller may be treated different than is child devices (i.e. USB card or disk). BIOS UEFI may allow firmware boot for a card but not for a device. No clue yet on this subject :-)

This way we may leverage UEFI standard to boot from a controller/device that is unknown to BIOS, either having Clover directly as the hardware firmware, or some glue that would boot off the unknown hardware and then launch Clover from a partition.

In  your case, as mentioned in my first post, you probably need to boot off the SATA drive in the first place, load bootloader, load kernel, and mount `/` from external USB drive. That external drive may not be visible to BIOS but it will be visible to OS Kernel. But you need to load that Bootloader and Kernel from a BIOS bootable device. This works for me. Also I know people can boot Bootloader and Kernel from USB then mount `/` from a controller, but their BIOS allows USB boot.

Yes I have some experience in BIOS hacking and necessary hardware like (de)soldering stuff and memory programmer in case something goes wrong, also can write utilities when necessary, so no problem here :-)

Morbius01, 2022-02-09:
I have read your OP and to your final question (Question: is it possible to flash Clover EFI Bootloader….), I do not have an answer.

However, if your goal is to install an operating system on your NVMe SSD (not visible from your BIOS/Motherboard) and then  boot from it, why don’t you just install Clover on a USB drive, and then install the OS on your NVMe ?

This will allow you to boot from your NVMe SSD through Clover, and it is why Clover was created !

And you won’t need to “hack” your BIOS, which could be a risky operation...

This is precisely what I recently did (Windows and Linux) and it works fine. See my post here :
[https://github.com/CloverHackyColor/CloverBootloader/discussions/491](url)

cederom, 2022-02-09:
Yes I have it working like this already :-) Boot from onboard SATA/USB then use `/` from NVM and it works. After upgrading motherboard to one that already has NVM support in BIOS even less problem because I can now boot directly from the NVM :-)

The question is more Research / Education / Experiment to learn how UEFI Firmware and NVM Controller Firmware and NVM Disk Firmware works :-)

More general question would sound: would that be possible to build Clover EFI Boot as UEFI Firmware that could boot directly from any PCI-E device :-)

This interesting project https://github.com/linux-nvme/nvme-cli allows to play with Firmwares.. no BIOS modification even necessary :-)