CloverHackyColor/CloverBootloader #497: Monterey 12.2 - Clover Bootloader v5144 - Required firmware update could not be installed.

SeanBlahovici, 2022-01-30:
Hello, first off I am not entirely sure this belongs here. Apologies if this has nothing to do with the Clover Bootloader.

So I tried doing a fresh install using a stable EFI (Tested on Mojave with updated drivers and kexts), but I am running into an issue during the installation of Monterey 12.2.

I tried following the steps here for people using Clover:
https://www.tonymacx86.com/threads/a-required-firmware-update-could-not-be-installed.316570/

But mostly the issue was apparently about AdviseFeatures not being enabled. I saw that AdviseFeatures appended the an "8" to ExtendedFirmwareFeatures and ExtendedFirmwareFeaturesMask in the config.plist/smbios.

However, I noticed that refreshing the firmware on my smbios added these ExtendedFirmwareFeatures with the bit correctly appended.

This is the relevant part of my plist:
    
```
        <key>ExtendedFirmwareFeatures</key>
        <string>0x8FD0FF576</string>
        <key>ExtendedFirmwareFeaturesMask</key>
        <string>0x8FF1FFF3F</string>
        <key>FirmwareFeatures</key>
        <string>0xFD0FF576</string>
        <key>FirmwareFeaturesMask</key>
        <string>0xFF1FFF3F</string>
```

However no matter what I try, I still get the message
A required firmware update could not be installed

Can anyone help? Much appreciated thank you!

SergeySlice, 2022-02-05:
The issue looks like you have wrong ExtendedFirmwareFeatures but you shown right value. May be your working config.plist is not same as you look?

SergeySlice, 2022-02-28:
No answer. Seems to be resolved.

closed 2022-02-28 by SergeySlice